dalathegreat/Battery-Emulator #620: Add Support for GoodWe A-ES Series Inverter

gegere, 2024-11-17:
### Expected Behavior

Currently on the [supported Inverter page](https://github.com/dalathegreat/Battery-Emulator/wiki#supported-inverters-list) the following support for Goodwe Inverters are listed. I am interesting in helping to confirm is it will work with the following model: 

[**GoodWe A-ES Series**](https://en.goodwe.com/a-es-series-split-phase-hybrid-solar-inverter-for-north-america)
(Americas Only) 5-9.6 kW | Split Phase | Up to 4 MPPTs | Hybrid inverter (HV)

Would this work with the same parameters as **GoodWe ET/BT & EH/BH & EHB**?

### Settings

- Software version: ``
- Battery used: `Tesla M3 Standard Range`
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `yes`
- DUAL_CAN: `yes`
- WEBSERVER: `yes`
- MQTT: ``

dalathegreat, 2024-11-17:
Should work just fine, BYD is on the [approved list](https://www.invertersupply.com/media/data/GW_Approved%20Battery%20Options%20Statement-EN.pdf) for this inverter! Let us know how it goes!

gegere, 2024-11-18:
Promising! Currently I am using ingenext Tesla Battery Controller and I am trying to resolve an issue charging on AC via the Tesla Mobile Charger OR j1772 Charger. The LilyGo RS485 Board will be arriving next week.

I am having an issue charging. The charge port will allow a charge a minute sometime up to 3 minuts, at 32 amps / 240VAC. 16 amps on each phase. Hopefully the **LilyGo RS485 / software** and this growing DIY battery community can help improve AC Charging and even DC/DC Fast Charging via the Tesla Charge Port.

Any tips OR thoughts on the situation?

[image]

Then PCS_hvChargeStatue: PCS_CHARGE_BLOCKED

[image]

dalathegreat, 2025-01-03:
Any progress on this testing @gegere ? :)

gegere, 2025-01-04:
Yahoo! I have not yet purchased the **GoodWe A-ES Series Inverter**. I am currently in the process of adding a radiator to cool the Tesla M3 PCS/Battery during AC/DC charging using the charge port, with plans to complete this by January 10th. The system's features will include management using ESP32 Arduino microprocessors over CANBus.

The next project will involve removing the **Ingenext Tesla Battery Controller**, as their support has been terrible, plus their system is closed source. 

I'll be setting up the Battery Emulator on a Lily Board instead. I have successfully installed the Battery Emulator on the Lily Board without any issues and have the web server up and running! Exciting!

I truly appreciate your thoughts, time, and energy in helping move this great project forward. I will continue to be an active contributor going forward!

gegere, 2025-06-24:
Outstanding! Everything out of the gate is working, so excited! Follow [these instructions](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Tesla-Model-S-3-X-Y#example-integration) made it feel like a morning breeze! 

CAN_INVERTER_MISSING
WARNING
6/24/2025, 2:03:47 PM
Inverter not sending messages via CAN for the last 60 seconds. Check wiring!

gegere, 2025-06-25:
The **CAN_INVERTER_MISSING** issue was resolved by installing a 120 Ohm resistor on the GoodWe Can connector. (where you plug the CAN_H & CAN_L to the inverter) 

An issue I didn't realize until now, is the GoodWe A-ES Series Inverter can't communicate with the battery. I can't run the inverter off the battery, which is odd because the Battery-Emulator is able to display all Canbus data from the TESLA_MODEL_3Y_BATTERY.

Tomorrow I am going to recompile with the following setting:
#define LFP_CHEMISTRY          //Tesla specific setting, enable this line to startup in LFP mode

gegere, 2025-06-30:
Happy to report a successful setup, this was the battery option I needed to select with the GoodWe SolarGo App.

[image]

closed 2025-06-30 by gegere

gegere, 2025-09-19:
@dalathegreat can this be added to the [upported-inverters-list](https://github.com/dalathegreat/Battery-Emulator/wiki#supported-inverters-list)?