dalathegreat/Battery-Emulator #625: max charge / discharge details for 2nd battery

pmajor021, 2024-11-19:
### Expected Behavior

Status page to display max charge and discharge current values for both batteries in case of dual battery setup.

### Actual Behavior

Status page only shows this for the first battery, making it difficult to compare similar values ​​between the two.

### Steps to Reproduce the Problem

It is on the status page.

### Settings

- Software version: `7.8.dev`
- Battery used: `BMW i3`
- Inverter communication protocol: `BYD over Modbus RTU`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- DUAL_CAN: `yes`
- WEBSERVER: `yes`
- MQTT: `yes`

dalathegreat, 2024-11-20:
Good suggestion @pmajor021 , fix is up in this PR: https://github.com/dalathegreat/Battery-Emulator/pull/628

dalathegreat, 2024-11-20:
Closing this as completed, now in 7.8.dev (Will be targeted for 7.8.0 release)

closed 2024-11-20 by dalathegreat