dalathegreat/Battery-Emulator #633: BMW i3 more info display value switched.

mbuhansen, 2024-11-24:
### Expected Behavior

In bmw i3 more info display dash soc and raw soc is switch around. display soc has to be highst value.
when bateery raw soc is 81% dispaly soc has to be 100%

In the last picture youyr can se it is switch around, it is in BMW I3.cpp

### Actual Behavior

### Steps to Reproduce the Problem

1.
1.
1.

### Settings

- Software version: ``
- Battery used: ``
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO, HW_STARK, Custom`
- CONTACTOR_CONTROL: `yes/no`
- DUAL_CAN: `yes/no`
- WEBSERVER: `yes/no`
- MQTT: `yes/no`

[image]
[image]
[image]

mbuhansen, 2024-11-24:
i have change the values in my file, and it is now show correct.

here is picture when i charge to 100%

[image]
[image]

dalathegreat, 2024-11-24:
Thanks for the report! Fix is up in https://github.com/dalathegreat/Battery-Emulator/pull/634 , closing this as completed!

closed 2024-11-24 by dalathegreat