dalathegreat/Battery-Emulator #638: Support single modules from volkswagen MEB platform

TheGreenJosip, 2024-11-25:
### Expected Behavior

I have a single 12S2P 6,85kWh module with ~44V out of a VW.ID3 battery pack connected with a Daly smart BMS supporting CAN/RS485 etc. How a single battery module from the respective platform can be used with the LilyGo?

### Actual Behavior

I have currently flashed the Lilygo using the BYD_MODBUS inverter communication protocol for the use with my Fronius SymoHybrid and RENAULT_TWIZY_BATTERY as battery type since this one matches most closely the single module I have. 

This will be my test setup and I'll post the actual behavior and output afterwards.

### Settings

BATTERY_WH_MAX 6850
BATTERY_MAXPERCENTAGE 9500
BATTERY_MINPERCENTAGE 500
BATTERY_MAX_CHARGE_AMP 1000
BATTERY_MAX_DISCHARGE_AMP 1000

- Software version: ``
- Battery used: `RENAULT_TWIZY_BATTERY `
- Inverter communication protocol: `BYD_MODBUS `
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- DUAL_CAN: `no`
- WEBSERVER: `yes`
- MQTT: `yes`

dalathegreat, 2024-11-25:
Hi,
To safely use the battery, you need a BMS that can talk CAN to the Battery-Emulator. The emulator can then do the final translation to the inverter.

At the moment we support the [RJXZS BMS](https://github.com/dalathegreat/Battery-Emulator/wiki/RJXZS-BMS) for custom batteries. This BMS can be used with any cell configuration.

If your Daly BMS supports CAN comm, we could add support for this also.

TheGreenJosip, 2024-11-25:
I would really appreciate if the CAN-support for the Daly BMS could be added and can provide any info needed as I have all of the hardware at home.

dalathegreat, 2024-11-25:
@TheGreenJosip Can you take a CAN log from your Daly BMS and post it here? 

[Instructions on how to log CAN](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging)

Download link to latest version of Battery-Emulator that has CAN logging in webserver https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/can-logging-webpage.zip

TheGreenJosip, 2024-11-26:
> @TheGreenJosip Can you take a CAN log from your Daly BMS and post it here?
>
>
>
> [Instructions on how to log CAN](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging)
>
>
>
> Download link to [...]

Good morning, 

Sorry I wasn't at my parents yesterday anymore but will provide the CAN-log today before midday as soon as I arrive. 

Thanks a lot for the quick help, I appreciate that very much 🙏

TheGreenJosip, 2024-12-02:
> @TheGreenJosip Can you take a CAN log from your Daly BMS and post it here? 
> 
> 
> 
> Download link to latest version of Battery-Emulator that has CAN logging in webserver https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/can-logging-webpage.zip

Hey @dalathegreat, I had covid and was in the bed the last days. May you sent me an updated version of the download link since this one does not seem to work properly? 

I'm at my parents tomorrow and could share the CAN-log of the Daly BMS. 

Thanks in advance 🙏

dalathegreat, 2024-12-04:
@TheGreenJosip It is expired since it merged to main.

Download main here: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/main.zip
[image]

TheGreenJosip, 2024-12-07:
Oh okay great I didn't notice it. Is the configuration also updated properly and can I find the right values for the single MEB-module?

Thank you 🙏

dalathegreat, 2025-05-25:
Closing as obsolete, new issue can be raised for adding Daly BMS if someone has CAN logs

closed 2025-05-25 by dalathegreat