dalathegreat/Battery-Emulator #640: BMW i3 battery with a Deye inverter

zsmaczko, 2024-11-25:
Dear contributors,

I am working on a project to integrate my BMW i3 battery with a Deye inverter Deye SUN 29.9-50K SG01HP3-EU-BM3 using your Battery Emulator. I noticed the wiki mentions the i3 battery uses the same CAN IDs as the BYD Battery-Box Premium HVS, requiring a separate CAN bus.

However, the LilyGo T-CAN485 board only has one CAN connector. Could you please advise on how to best address this?

I have considered these options:

Separate CAN bus with an additional CAN interface: This seems like the most straightforward solution, but it adds to the hardware cost. This also means I have to modify the software.
CAN multiplexer: This could potentially allow me to use a single LilyGo board, but I'm unsure about the complexity and compatibility.
Modifying the Battery Emulator software: I am willing to explore this if it's feasible, but I would need guidance on how to approach it.
I would appreciate any insights or recommendations you might have.

Thank you for your time and for developing this excellent project!

### Settings

- Software version: `7.8`
- Battery used: `BMW i3`
- Inverter communication protocol: `CAN`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- DUAL_CAN: `no`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2024-11-25:
Hi @zsmaczko 

This can be solved in three ways:

    You can use [Double Lilygo's](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki/Double-LilyGo)
    You can [add an isolated MCP2515 CAN channel](https://github.com/dalathegreat/BYD-Battery-Emulator-For-Gen24/wiki/Dual-Can-(MCP2515-add%E2%80%90on))
    You can use the [Stark CMR](https://github.com/dalathegreat/Battery-Emulator/wiki/Stark-CMR) board

I have updated the BYD wiki. Personally I suggest going with the 5€ MCP2515 chip.

zsmaczko, 2024-11-27:
Dear @dalathegreat,

Thank you so much for your quick reply and the detailed suggestions! I especially appreciate you taking the time to update the wiki with this information.

I decided to go with the first option and add an isolated MCP2515 CAN channel. This solution worked perfectly for me, and I haven't (don't need to) explore the other options.

Thanks again for your help and for developing such a valuable tool for the community.

Sincerely,
Zsolt

closed 2024-11-27 by zsmaczko

DIYBMSInverter, 2025-01-07:
Hey zsmaczko,

We are currently trying to connect our battery to a Deye SUN 29.9-50K SG01HP3-EU-BM3 like you did. We are transferring the correct battery values but receive the error message 'F58 - BMS communication fault'. Do you have any experience with that?

Kind regards

dalathegreat, 2025-01-07:
> Hey zsmaczko,
> 
> We are currently trying to connect our battery to a Deye SUN 29.9-50K SG01HP3-EU-BM3 like you did. We are transferring the correct battery values but receive the error message 'F58 - BMS communication fault'. Do you have any experience with that?
> 
> Kind regards

New issues should be created, and old closed ones not hijacked!