dalathegreat/Battery-Emulator #646: Emulator standalone?

jakub8109, 2024-11-27:
good day
I need to use the emulator without the inverter - is that possible?

I also need such a possibility that when the emulator determines that the battery is already charged, it will e.g. activate the pin on which the relay will be.

smaresca, 2024-11-28:
We do support two chargers - the Nissan Leaf PDM and the Chevrolet Volt (Opel Ampera) charger unit, for battery commissioning and emergency charging, so that is possible.  I don't believe we have implemented any contactor/relay control through in the chargers.  

Could you describe your use case a bit more?

jakub8109, 2024-11-28:
thanks for the answer. I have batteries from hyundai ioniq5. the batteries are 750-800v and will be charged directly from the pv panels. in order not to overcharge them I would like the battery emulator to give info on pin io5 to disconnect charging.

mvgalen, 2024-12-04:
How is the load connected to your battery?
To implement your contactor control, I am wondering if adding a 'CONTACTOR_CONTROL' inverter would be an option. With some configurable output pins. So instead of a RS485/CAN protocol it would just speak GPIO.

jakub8109, 2024-12-04:
an option without an inverter would be great. when the battery reaches 100%, give a high signal to the pin. charging would take place straight from the PV panels.

mvgalen, 2024-12-04:
> charging would take place straight from the PV panels.
But how do you plan to use the power from the battery?

jakub8109, 2024-12-04:
the battery is connected together with the PV. when there is sun, the batteries are charged. the emulator is supposed to monitor the battery along with the BMS, when it is charged it is supposed to give information on the pin - then the voltage from the PV will be redirected to the inverter. when the sun sets, my device will release battery power. therefore, the contactors must be constantly connected.

mvgalen, 2024-12-05:
Let me get this straight; you plan to connect the battery, inverter and PV panels in parallel? So, discharging the battery through the inverter (which I guess is not a hybrid inverter?) ?

jakub8109, 2024-12-05:
yes, that's right.

nagyrobi, 2024-12-05:
I think it's not a too well thought-out idea...

mvgalen, 2025-01-02:
I think he consensus here is: 
- Hybrid inverters are suited for this scenario
- Running battery and inverter in parallel is not a good idea. You will completely ruin the MPPT behavior of the inverter.
- Discharging a battery through the PV input of an inverter is also not a good idea, the inverter will pick an MPPT point where it maximises powerdraw from the battery

If you want further discussion please reopen the issue.

closed 2025-01-02 by mvgalen

jakub8109, 2025-01-11:
please add an option without an inverter.