dalathegreat/Battery-Emulator #651: BMW 530e PHEV G30 Battery log

Refa2424, 2024-12-01:
### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

1.
1.
1.

### Settings

- Software version: ``7.8.0
- Battery used: ``5 Series G30 PHEV (SP06)-2018
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO
- CONTACTOR_CONTROL: `yes/no`
- DUAL_CAN: `yes/no`
- WEBSERVER: `yes/no`
- MQTT: `yes/no`
- 
[CoolTerm Capture (Untitled_0) 2024-12-01 09-04-10-050.txt](https://github.com/user-attachments/files/17968674/CoolTerm.Capture.Untitled_0.2024-12-01.09-04-10-050.txt)

lenvm, 2024-12-12:
@Refa2424, can you please add a description to this issue, such that it is clear what needs to be done to resolve this issue?

Refa2424, 2024-12-13:
basically, to try to find a solution to integrate the said battery with Lilygo by using communication with the SME module inside the battery.
some solutions exist, but they are not complete because they do not use the SME module.
Some things have already been done, we have a slight progress.
as directed by Dalathegreat to add this issue here so we don't forget since the same thing is being discussed on Discord.

wjcloudy, 2025-01-06:
Just to add, we have read only working, but need can-logs from a working BMW PHEV (3,5,x5, etc) PT-CAN2 bus to work on balancing/iso measurement/contactors. If anyone has one of these, please get in touch!

wjcloudy, 2025-01-19:
Closing as Beta support added in #804 (which needs manual contactor control). We still need logs from an intact vehicle to finish a full integration.

closed 2025-01-19 by wjcloudy