dalathegreat/Battery-Emulator #664: Off Grid - Contactor Control via GPIO

GoSmartEV, 2024-12-10:
What

If you are off grid, and the battery is not turned on and there is no sun, then the inverter will be turned off “ because only the battery can turn on the inverter”

Why

In the description to “ Contactor Control via GPIO “ it says that 
“ Before the contactors turn on, both Inverter and Battery needs to give OK ✅ signal. This can be verified via the Webinterface. In this screenshot, battery is preventing startup “

How

but how will the inverter send something via CAN  if the battery does not turn on and does not supply power to the inverter

dalathegreat, 2024-12-11:
I think this was an issue with overloaded pins for contactor control? Looks like you got it working on Discord? :)

GoSmartEV, 2024-12-11:
Yes, I tried to delete this post, but I couldn't find how

but for future, you need to add a description of how to find problems, because it is very deceptive

as i got confused, looking at the webpage, and saw the ❌

need to add that if you don't have control and there are no errors, try disconnecting the wires from GPIO pins, and after restarting the lilygo if u see ✅ this means you have a over current on the GPIO pins and u must use a other SSR

dalathegreat, 2024-12-11:
Added section on overloaded GPIO pins to wiki: https://github.com/dalathegreat/Battery-Emulator/wiki/Contactor-Control-via-GPIO-pins#overloaded-gpio-pins

Hopefully this info helps someone else also! Closing this! :)

closed 2024-12-11 by dalathegreat