dalathegreat/Battery-Emulator #670: SUNGROW CAN

itsallanetwork, 2024-12-13:
### Expected Behavior

Inverter should talk CAN to lilygo

### Actual Behavior

I have tried MCP2518 board, reverted to Dual lillygo, thinking it may be the board (I had 2 lillygo's on hand). Connection to the 2021 Model 3 (M3) Tesla battery is perfect in everyway, I am having troulbe connecting to the Sungrow SH5.0RS inverter.

Tesla Battery Situation:
On software 7.9 or 7.6 (dev for Sungrow):
I have logged comms over R364 to the other lilygo and I am seeing the battery data come over accurately. 
I can use all of the buttons on the webserver and all function correctly.
Battery lilygo LED is GREEN 
No issues.

Sungrow SH5.0RS situation:
Inverter on OLDER Software:
Module Name	      Module Current Version
SDSP		SUNSTONE-H_04011.02.03
MDSP		SUNSTONE-H_03011.02.33
LCD		       SUNSTONE-H_01011.02.44
AFCI		       AFCI_06002.01.10

Sungrow changed their Support for BYD on June 2024. I am not sure if this effects the inverter's reporting to the Cloud software (isolarcloud.com). I very much doubt it, but it cant be ruled out.

Inverter Lillygo (IL) is seeing Battery Data from Tesla Battery and reading it correctly. 
All voltages are good on the IL, 
IL definitely sends CAN messages (I can see them on 7.9). 
IL is not receieving any CAN from Inverter

When Inverter Lilygo starts, in order, the LED:
1) flashes green
2) flashes red
3) flashes green
4) flashes red and stays red with FAULT below:

FAULT Displayed in EVENTS:
CAN_TX_FAILURE
ERROR
12/10/2024, 9:41:48 PM
COUNT = 3
DATA = 0
ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!

Then serial output:
22:55:11.102 -> CAN failure! Check wires
22:55:11.102 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:11.568 -> CAN failure! Check wires
22:55:11.568 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:12.068 -> CAN failure! Check wires
22:55:12.068 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:12.590 -> CAN failure! Check wires
22:55:12.590 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:13.092 -> CAN failure! Check wires
22:55:13.092 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:13.568 -> CAN failure! Check wires
22:55:13.568 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:14.092 -> CAN failure! Check wires
22:55:14.092 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:14.608 -> CAN failure! Check wires
22:55:14.608 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:15.089 -> 60000 SerialDataLink-Receiver - NewData :33   Errors : 0
22:55:15.089 -> Values from battery: 
22:55:15.089 -> SOC: 8610 SOH: 9900 Voltage: 3890 Current: -2 Capacity: 82000 Remain cap: 70600 Max discharge W: 60000 Max charge W: 0 BMS status: 4 Power: -70 Temp min: 250 Temp max: 255 Cell max: 4074 Cell min: 4068 LFP : 0 Battery Allows Contactor Closing: 1 Inverter Allows Contactor Closing: 1
22:55:15.089 -> CAN failure! Check wires
22:55:15.089 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:15.615 -> CAN failure! Check wires
22:55:15.615 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
22:55:16.089 -> CAN failure! Check wires
22:55:16.089 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
[... 264 lines cut ...]

#endif  // WEBSERVER
// MQTT
#ifdef MQTT
const char* mqtt_user = "BlackBox24";      // Set NULL for no username
const char* mqtt_password = "ToyotaSupra1994";  // Set NULL for no password
#endif                                   // USE_MQTT
#endif                                   // WIFI

#ifdef EQUIPMENT_STOP_BUTTON
// Equipment stop button behavior. Use NC button for safety reasons.
//LATCHING_SWITCH  - Normally closed (NC), latching switch. When pressed it activates e-stop
//MOMENTARY_SWITCH - Short press to activate e-stop, long 15s press to deactivate. E-stop is persistent between reboots
volatile STOP_BUTTON_BEHAVIOR equipment_stop_behavior = LATCHING_SWITCH;
#endif

/* Charger settings (Optional, when using generator charging) */
volatile float CHARGER_SET_HV = 384;      // Reasonably appropriate 4.0v per cell charging of a 96s pack
volatile float CHARGER_MAX_HV = 420;      // Max permissible output (VDC) of charger
volatile float CHARGER_MIN_HV = 200;      // Min permissible output (VDC) of charger
volatile float CHARGER_MAX_POWER = 3300;  // Max power capable of charger, as a ceiling for validating config
volatile float CHARGER_MAX_A = 11.5;      // Max current output (amps) of charger
volatile float CHARGER_END_A = 1.0;       // Current at which charging is considered complete

+++++++++++++++++++++++++++++++++++++ END +++++++++++++++++++++++++++++++

itsallanetwork, 2024-12-13:
--------------- 7.6 DEV  (Sungrow CAN)  ------------------

12:10:38.920 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:10:38.952 -> configsip: 0, SPIWP:0xee
12:10:38.952 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:10:38.952 -> mode:DIO, clock div:1
12:10:38.952 -> load:0x3fff0030,len:4832
12:10:38.952 -> load:0x40078000,len:16460
12:10:38.952 -> load:0x40080400,len:4
12:10:38.952 -> load:0x40080404,len:3504
12:10:38.952 -> entry 0x400805cc
12:10:39.453 -> __ OK __
12:10:39.453 -> EEPROM was initialized for event logging
12:10:39.453 -> head: 0, tail: 0
12:10:39.453 -> No events in log
12:10:39.495 -> SERIAL_DATA_LINK_RECEIVER selected
12:10:39.495 -> Info: The board was reset via software, webserver or OTA. Normal operation
12:10:39.580 -> Creating Access Point: Battery Emulator
12:10:39.580 -> With password: 123456789
12:10:39.580 -> Access Point created.
12:10:39.580 -> IP address: 192.168.4.1
12:10:39.580 -> Connecting to Wi-Fi...
12:10:39.580 -> Info: Wifi connected.
12:10:39.580 -> Wi-Fi connected.
12:10:39.580 -> IP address: 192.168.1.201
12:10:39.618 -> Wi-Fi Got IP.
12:10:39.618 -> IP address: 192.168.1.201
12:10:39.887 -> CAN failure! Check wires
12:10:39.887 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:40.397 -> CAN failure! Check wires
12:10:40.397 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:40.907 -> CAN failure! Check wires
12:10:40.907 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
...
12:10:52.380 -> CAN failure! Check wires
12:10:52.380 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:52.880 -> CAN failure! Check wires
12:10:52.880 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:53.380 -> CAN failure! Check wires
12:10:53.380 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:53.880 -> CAN failure! Check wires
12:10:53.880 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:54.426 -> CAN failure! Check wires
12:10:54.426 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:54.927 -> CAN failure! Check wires
12:10:54.927 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:55.380 -> CAN failure! Check wires
12:10:55.380 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:55.880 -> CAN failure! Check wires
12:10:55.880 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!
12:10:56.380 -> CAN failure! Check wires
12:10:56.380 -> ERROR: CAN messages failed to transmit, or no one on the bus to ACK the message!

itsallanetwork, 2024-12-13:
[image]

Attached - Battery Discharges on the Inverter. The Inverter shows SOC, but no Charging (MAYBE due to the LilyGo Seeing a fault).

mathl79, 2024-12-13:
in your user_settings.h the #BYD_CAN option is anabled and the SUNGROW_CAN option is not present. 
Did you pull the sungrow branch?

mathl79, 2024-12-13:
Again, are you sure, you are running the "feature/sungrow_can" branch? 
It is based on 7.9 according to the readme.md.

dalathegreat, 2024-12-13:
@itsallanetwork 
I have updated the PR with the latest main (v8.0.dev). Can you try this version, and take som [CAN logs](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-logging) while running the sungrow protocol?

https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/sungrow-can.zip

itsallanetwork, 2024-12-13:
Updated the inverter and the battery with 8.0.dev

Same TX-Error:
[image]

Can log, only seeing TX from the Lillygo, as follows:

[canlog_Inverter-CANLOG-14-12-2024-0850h.txt](https://github.com/user-attachments/files/18132647/canlog_Inverter-CANLOG-14-12-2024-0850h.txt)

itsallanetwork, 2024-12-13:
This is the REBOOTED and captured the CAN LOG Immediately and it shows RX!!!

[image]

itsallanetwork, 2024-12-13:
screen captures of the Inverter CAN LOG showing RX

[image]
[image]
[image]

itsallanetwork, 2024-12-13:
It seems the RX4200 message coincides with the LED:
1) flashes green
2) flashes red
3) flashes green
4) flashes red and stays red with TX FAULT

Sorry for the tonne of posts, just giving as much information as possible.

itsallanetwork, 2024-12-14:
@mathl79 - Are you running the latest Sungrow software? Is that what this CAN BUS was written on, because I am not running the latest software on my Inverter. The reason Im running older software is in June 2024 Sungrow AU discontinued support for BYD Batteries and I am concerned that means their latest software will not recognised BYD CAN.

mathl79, 2024-12-14:
My SH15T Inverter is running the latest firmware (Europe) and BYD + Pylon batteries are still supported. For this inverter type also the manual states that only sungrow batteries are supported. 
Maybe this is only some kind of brand selling tactics ?!?

dalathegreat, 2024-12-14:
@itsallanetwork Message 0x4200 indicates that the Sungrow inverter is trying to establish a Pylon Battery connection. It looks like your inverter could work with the protocol:

#define PYLON_CAN        //Enable this line to emulate a "High Voltage Pylontech battery" over CAN bus

Try this!

itsallanetwork, 2024-12-14:
@dalathegreat I have I have loaded PYLON_CAN. The Inverter Lilygo is Green/happy, but the Inverter is not reading the battery data correctly:

[image]

dalathegreat, 2024-12-14:
@itsallanetwork  Looks like the Pylon is almost working on your inverter. The Pylon protocol is heavily configurable, next steps are to adjust some of the settings inside the Pylon protocol file: Battery-Emulator/Software/src/inverter/PYLON-CAN.cpp

[image]

I suggest testing both switching to SEND_1 instead of SEND_0, and also another test with enabling SET_30K_OFFSET , this might explain why some values are corrupted.

I think you are very close to having the first Sungrow working with Pylon. Once you do, we should document which settings work, and add it to the Wiki

itsallanetwork, 2024-12-14:
@dalathegreat Thanks again! I will try all of that and document results

Once we get this old inverter software working I have a second SH5.0RS which I have updated to the latest software. I will swap physical inverters and test again. I would really like to help get the latest software working for everyone, Sungrow stopped support for the SH10RS inverters first. I am guessing they will need SUNGROW_CAN.

itsallanetwork, 2024-12-15:
@dalathegreat I tried a bunch of settings with PYLON_CAN. None seemed to work. None showed any SOC change:
[image]

itsallanetwork, 2024-12-15:
@dalathegreat  I went back to BYD_CAN on version 8.0 and it seems to be working. BUT I dont **seem** to be getting a CAN signal back from the inverter (on CAN LOGGING):
[canlog_02-18-59.txt](https://github.com/user-attachments/files/18141392/canlog_02-18-59.txt)

[image]

[image]

VERY ODD.

itsallanetwork, 2024-12-16:
10am, status is working:

[image]

dalathegreat, 2024-12-16:
Could you please write down your exact model number of inverter, firmware version on the inverter, and the settings used for the emulator? That will help the next person with this type of Sungrow!

itsallanetwork, 2024-12-16:
@dalathegreat Yes of course! 

Sungrow SH5.0RS
Inverter on OLDER Software:
Module Name Module Current Version
SDSP SUNSTONE-H_04011.02.03
MDSP SUNSTONE-H_03011.02.33
LCD SUNSTONE-H_01011.02.44
AFCI AFCI_06002.01.10

I beleive the issue was the BYD-CAN.cpp checking for the Inverter and then going to fault.  8.0.dev does not have this issue. 7.6.dev does. When I compare both files, 7.6.dev and 8.0.dev, I see quite a few differences:

[image]
[image]
[image]
[image]
[image]

The other problem is, these inverters will not take an older software, even if you try to push it to them without Wifi/cloud connection. They deny the older software, no way to go backward in version. More testing on the newer software is needed to cover most users.

dalathegreat, 2024-12-17:
Thanks, I updated the Wiki with the info on this older AU version: https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Sungrow

itsallanetwork, 2024-12-27:
Hi @dalathegreat I have been running on BYD with no issues for weeks now. I would like to help develop the Sungrow CAN.

I have completed the following:

Swapped over to sungrow can (only by unediting the line SUNGROW_CAN and commenting out BYD_CAN). This way we are only testing the difference between the two CAN protocols.  This threw up the same error in the EVENTS "no one to ACK the message". 

I beleive the SUNGROW does not ACK any messages. If we tested this by removing that as a requirement in the SUNGROW_CAN, then the lillygo would not 'fault' and we might get it working.

I am happy to test this, as I have the Tesla battery and it's contactors shut when the other lilygo is not passing messages. I will also test at 80% battery.

I will swap inverters, to the one with the NEW Sungrow Software today. Then test both BYD and SUNGROW.

itsallanetwork, 2024-12-28:
I am now running the latest SUngrow software on the SH5.0RS, with the prior working 8.0.dev software, the inverter lilygo is showing 'fault'/red. Event shows:

[image]

The battery seems to be working though (I will report back after some time):

[image]

Im not sure what to do about the RX issues causing a fault on the Inverter LilyGo. Help?

itsallanetwork, 2024-12-28:
When I try Sungrow_CAN I get a the same Event error "CAN-TX-ERROR".

With the CAN Logging showing:

[image]

Battery shows up 'normally' on the Inverter:

[image]

GENIUS input required please @dalathegreat  :)

dalathegreat, 2024-12-28:
@itsallanetwork 
    case 0x00004200:  //Only sent by SH15T (Inverter trying to use Pylon CAN)
      datalayer.system.status.CAN_inverter_still_alive = CAN_STILL_ALIVE;
      break;
    case 0x02007F00:  //Only sent by SH15T **(Inverter trying to use Pylon CAN)**
      datalayer.system.status.CAN_inverter_still_alive = CAN_STILL_ALIVE;

Looks like your inverter is expecting a Pylon protocol!

itsallanetwork, 2024-12-28:
Thanks @dalathegreat, I will try Pylon again.

itsallanetwork, 2024-12-28:
So I put PYLON_CAN on the Inverter LilyGo:
1) The lilygo went green, 
2) The SH5.0RS didnt recognise the battery or the SOC
....   I left it for 30mins, no change
3) I put BYD_CAN on the Inverter LilyGo 
4) Everything is working perfectly.

It _seems_ the inverter wants to see 'hello' with the PYLON CAN and then actually _work_ with the BYD CAN....   

That is my **guess** from a process of elimination....

itsallanetwork, 2024-12-28:
Here is the CAN LOG after I have run the PYLON CAN, it is absent of any RX:

[canlog_22-39-34.txt](https://github.com/user-attachments/files/18268026/canlog_22-39-34.txt)

dalathegreat, 2024-12-30:
So in order to work with the new software, you start with Pylon CAN and then go to BYD Can?

itsallanetwork, 2025-01-05:
@dalathegreat - It seemed that way, but then i just had a dangerous situation where the inverter kept sending charge to the battery and only the failsafes stopped the battery from overcharging. I have had eye trouble and could not monitor, nor work on it recently.

dalathegreat, 2025-01-05:
@itsallanetwork This is most likely this bug, that we have in the Wiki for Sungrow;

"ℹ️ If you are using only a single LilyGo board, Sungrow has a nasty software bug. Incase the LilyGo board is disconnected while the system is running (wire break or hardware failure), the inverter will still charge/discharge the battery! This is because the Sungrow inverter sees the automotive CAN messages as the system still being alive. This can lead to a dangerous situation where the battery is over/under-charged. This can be avoided by using a dedicated CAN channel for the Inverter!"

dgwynne, 2025-04-13:
> So in order to work with the new software, you start with Pylon CAN and then go to BYD Can?

we just tried this with code from the main branch and it seemed to work.

for the sake of clarity, we were trying to use SUNGROW_CAN against an SH5.0RS running currently firmware, but had no success. we were plagued by CAN TX errors. out of desperation we tried using the PYLON_CAN code, and then built and upgraded the lilygo to use BYD_CAN. the inverter was able to see SoC and was happily charging/discharging the battery fine after that.

turning everything (inverter and battery) off and then on again with BYD_CAN still in place didn't work. we got error 714 on the inverter. it's possible there's a physical comms issue.

we also haven't seen if the inverter will overcharge the battery or not yet.

jesseklm, 2025-08-11:
@joshbohmer Nice setup! Logs of CAN communication in both directions would be helpful, because it's easier to switch from BYD and much of the reverse engineering has already been done.

I'm currently working on implementing the Sungrow protocol for the SH20T with a custom battery.

joshbohmer, 2025-08-11:
I've just gone to 2 x SH15T's with 2 towers each. I'll hook up the CAN man in the middle then capture some logs for you.

Might take me a bit to get the equipment and set it up tho. I'd be very interested in custom batteries on Sungrow inverters, happy to help if I can.

aferrara, 2025-09-22:
Hi folks, I have been working on the Sungrow battery protocol for a while now and have made a decent about of headway. I have a single stack SBR096 at one site connected to a SH10RS at one site, and a SH10RS with no battery at another.

I would be really interested in some dumps of a multiple stack setup as I am pretty confident I know where the difference is.

I have most of the 0x700 and 0x710 series of messages decoded and I have a hypothesis of some of the missing messages.

It looks like the 0x500 -> 0x506 are the same as 0x700 -> 0x706 with the exception of 0x704 where one byte has the inverse sign of the same value.

aferrara, 2025-09-24:
Does anyone want to compare notes?

@theunknown86 @jesseklm