dalathegreat/Battery-Emulator #672: Fronius Gen24 Hybrid setup (BYD emulator)

windydrew, 2024-12-14:
If I purchase a Fronius Gen24 10kw hybrid that is set up for a BYD HV battery, can I use similar code to allow connection to a Leaf battery? 

- Software version: ``
- Battery used: ``2016 Leaf 40kW
- Inverter communication protocol: ``RS485 Modbus 
- Hardware used for Battery-Emulator: `Custom`
- CONTACTOR_CONTROL: `yes`
- DUAL_CAN: `yes`
- WEBSERVER: `yes`
- MQTT: `yes`

dalathegreat, 2024-12-14:
Yes, that is what I use, and it has been supported ever since the v1.0.0 release of the Battery-Emulator!

closed 2024-12-14 by dalathegreat

windydrew, 2024-12-15:
So it's been tested with Fronius Gen24 inverters?

dalathegreat, 2024-12-15:
Yes @windydrew , all batteries listed under the compatible list work with the Fronius Gen24 https://github.com/dalathegreat/Battery-Emulator/wiki#supported-batteries-list

windydrew, 2024-12-15:
Thanks, hadn't stumbled across that page yet. Just starting my research. I
also have an operational (salvage title) 2016 Leaf that I would be able to
test on. I see that V2h is testing, so I can help with that since I have a
Chademo cable to work with.