dalathegreat/Battery-Emulator #677: CAN issues with the lilygo and Volvo XC40 - Polestar 2	78kWh

GoSmartEV, 2024-12-16:
When 12V power is applied to the battery and the LilyGo board, there is occasionally an issue where the CAN bus and the transceiver in the BMS control unit do not "wake up," making communication impossible. So far, I’ve found only one solution to this problem — adding an external MCP2515 module and configuring it for communication with the BMS. In this setup, everything works smoothly and without any issues, ensuring reliable data transmission over the CAN bus.

[image]

lenvm, 2024-12-20:
For our understanding, can you please clarify what setup does not work and what setup does work?

Some questions:
- You mention "When 12V power is applied to the battery and the Lilygo", is this the case in both the working setup and the non working setup?
- Have you tried a setup where you are powering the Lilygo with power via USB?

GoSmartEV, 2024-12-21:
> For our understanding, can you please clarify what setup does not work and what setup does work?
> 
> Some questions:
> 
> * You mention "When 12V power is applied to the battery and the LilyGo", is this the case in both the working setup and the non working setup?
> * Have you tried a setup where you are providing the LilyGo with power via USB?

Yes and yes, there is a problem exactly with the lilygo this is 100% info

dalathegreat, 2025-01-03:
Closing this issue as completed, should be fixed now with PR https://github.com/dalathegreat/Battery-Emulator/pull/744

closed 2025-01-03 by dalathegreat