dalathegreat/Battery-Emulator #678: HYUNDAI KONA 40KWH BATTERY PACK THE EMULATOR SHOWS HIGH VOLTAGE ABOUT 6553V

anufahim, 2024-12-17:
### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

1.
1.
1.

### Settings

- Software version: ``7.9.2
- Battery used: ``HYUNDAI KONA 40KWH
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO, HW_STARK, Custom`
- CONTACTOR_CONTROL: `yes/no`
- CAN_ADDON: `yes/no`
- WEBSERVER: `yes/no`
- MQTT: `yes/no`
[image]
[image]

dalathegreat, 2024-12-17:
Sorry your battery might have internal issues, see the troubleshooting section: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Kia-Niro---Hyundai-Kona-64-kWh#troubleshooting

dalathegreat, 2024-12-28:
@anufahim Were you able to find the problem with the battery?

closed 2024-12-29 by anufahim