dalathegreat/Battery-Emulator #679: Webs server display of "Contactors controlled by emulator, state:" not working correctly

davidhgriffiths, 2024-12-18:
### Expected Behavior

When the contactors open after an inverter request the web server should display the OFF state

### Actual Behavior

When the contactors open after an inverter request the web server continues to display the  ON state

I think the bug is caused the absence of the following line at about line 128 of web server.cpp

datalayer.system.status.contactors_engaged = false;

currently datalayer.system.status.contactors_engaged only gets set to false if contractor opening is requested from the web page, not if it is requested by the inverter or estop

mvgalen, 2025-01-02:
Just so that I understand the issue correct: You mean that currently if an inverter request opening of contactors it does not show on the webpage? For which inverter have you seen this?

davidhgriffiths, 2025-01-04:
Yes your understanding is correct - if the inverter requests opening of the contractors the contractors open but the web page continues to show them closed.  The inverter is a Solax X3 G4. With the change I suggested it works correctly.

dalathegreat, 2025-01-20:
Thanks for the report @davidhgriffiths , I have issued a fix for this in PR https://github.com/dalathegreat/Battery-Emulator/pull/812

dalathegreat, 2025-01-23:
Closing as completed, now in main, target release 8.3.0 🥳  Thanks again for the report!

closed 2025-01-23 by dalathegreat