dalathegreat/Battery-Emulator #680: Lilygo constantly rebooting

tjjq44, 2024-12-19:
After flashing the board, Wifi AP isn't activated and I see this constantly looping on serial monitor...

ELF file SHA256: b5fd3e836

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8

assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

Backtrace: 0x400827d1:0x3ffb3010 0x4008e651:0x3ffb3030 0x40094a46:0x3ffb3050 0x40103903:0x3ffb3180 0x40103a7d:0x3ffb31a0 0x400dd6d4:0x3ffb31c0 0x400e3446:0x3ffb3210 0x400d933d:0x3ffb3230 0x400d326e:0x3ffb3280 0x4008f39a:0x3ffb32a0

### Settings

- Software version: `7.92`
- Battery used: `None (for now)`
- Inverter communication protocol: `CAN (Sunny Boy Storage 3.7)`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `yes`
- CAN_ADDON: `yes`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2024-12-19:
HI @tjjq44 , could you attach your configuration files (USER_SETTINGS.h and .cpp ), but be sure to remove Wifi login info from them!

dalathegreat, 2024-12-19:
Issue seems repeatable after updating ESP32 to v 3.1.0 

[image]

@tjjq44 you might be able to continue by downgrading ESP32 to v3.0.3 or similar older version.

tjjq44, 2024-12-19:
Thank you! This time it boots ok but... in both mode (AP and normal wifi) I can't log in the webserver. 

In AP mode I "see" the network "battery emulator" but connection is quickly lost with my iphone. In normal mode, I even can't ping the board (ssid, pw ans IP are correctly set). 

In both cases, the led flashes in green slowly for about 1 minute, the it goes still red. On serial monitor this message pops from time to time : "wifi:sta is connecting, return error".

Maybe I get these issues because there is no battery and no inverter yet connected to the lilygo?

[tjjq44-bat-emu.zip](https://github.com/user-attachments/files/18201543/tjjq44-bat-emu.zip)

dalathegreat, 2024-12-19:
@tjjq44 I fixed the root cause in this PR: https://github.com/dalathegreat/Battery-Emulator/pull/681

Please try it out, you can download the PR via this link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/bugfix/esp32-3.1.0-crashfix.zip

dalathegreat, 2024-12-20:
The fix has been merged into main (8.0.dev). Closing this issue as completed, thanks for the report!

closed 2024-12-20 by dalathegreat

tjjq44, 2024-12-22:
Thank you very much, everything is fine now. Will test it with a Ion battery when I get it in a few months!