dalathegreat/Battery-Emulator #692: Inception of New Project - DIY Home On/Off Grid solution using Tesla EV battery No Solar or wind.

KenG-TeslaAU, 2024-12-22:
Hi & Best regards to the Community 
I may not be in the right place, please be nice to me its my first time.  If I'm not in the right place can someone direct me were to go. 
Need Initial direction and advice
Outline
We are in Australia, live next to a National park lots of trees and bush. Solar and wind not possible.
Our energy, 230V 50 Hz Single phase.  Price per Kw varies every 5 minutes in low demand times can be as cheap as -27 cents (that is not a typo) to $20.56 cents/ Kw if there is a price spike. Whilst these prices are the extremes more commonly 10-20 cents/Kw in low usage times and 40-80 cents/Kw in high usage times.  Price spikes are common in mid summer and mid winter. These price spikes can continue for as much as 5 hours on very hot or very cold days. 

In addition to price spikes, there is also a Demand Fee imposed by the owner of the poles and wires network.  This fee is based on your highest monthly Kw usage during any 30 minute block through the hours 3 pm - 9pm daily.  Turning on an air-conditioner for 30 minutes during this time, just once, can add over $70 to a monthly bill even if there is no price spike. The demand period really limits usage as the price is simply outrageous. 

In some aspects our energy supplier has an app so we know what to expect of pricing, but during Demand Period and price spikes which often occur simultaneously, event the smallest usage can be very expensive so we basically switch off everything until the grid settles. 

We could elect have a flat rate per Kw fee about 35 cents but we can't remove the Demand Fee so no real advantage.  A another option is a much more expensive 45-50 cents flat rate per Kw but suppliers are becoming less will to do these for smart meter owners.  
Why don't I do the 40-50 cents flat tariff instead?  I have an EV and most times I can charge it at low times for 2 cents/Kw, so I'm not crying poor, and,  I do save a lot of money most times by using an automated charging app that only charges the car when prices are low. 

Anyway enough of the story.  
My plan is to use an EV battery about 80 Kw, likely to be a Tesla, to buy energy cheap and store it.  
During price spikes I plan to switch from the grid to the battery and I would do the same during the Demand periods.   
I have no intention to export back to the Grid if I did this I would need to choose a very expensive inverter which is must be approved by the AU Regulator an must be closely monitored.  These approved inverters are more about solar power inputs and supply regulation so they come at a price.  I don't see the point of paying for all the technology related to grid export solar panel monitoring and pure sine way phased mated energy exports.   

My needs are different.  I charge a battery and switch the grid OFF every day using a timer.  This allows me to avoid the Demand period.  When there are price spikes that occur outside the demand period I get txt messages warning me so I can manual switch off the grid and avoid these as well.     

Currently I can't use the Air-conditioner or other power hungry devices during demand periods or price spikes.   Once I have a battery I can begin to use my air-conditioner and oven when ever I choose but still take advantage of the cheap times to charge the battery and the car.  

Is this an economical decision?
Probably not, if there is a payback it has a long tail (one that I may never see) but,  it is immediately cheaper than other options and gives me my choice back when I want to use creature comforts and appliances.    

What I need
Help to choose a hybrid? single phase inverter less focused on solar

Cabling needs to the sub-board in the battery room 

Other Cable install (wiring needs)
- Power supply for the LillyGo from where
- Power supply for the Tesla EV Battery from where
- Power supply for the contactors from where
- Cable CAT 6 needed?
- Any other cable that would have been a good idea to have?

 **Help to determine the health of an EV battery removed from a wreck.** 

**Help to choose a hybrid? single phase inverter less focused on solar**
I would need about 10 -12Kw output from the battery.  That is enough to run an air-conditioner, oven and general smaller loads.  I would need it to switch to batteries and visa versa in milliseconds.  It would not be good for a huge EMF back wash to be created every time I do a change over and I'm not keen to reset the clocks and wait for the internet to return. 

**I'm not charging the car from the battery.**    
 
**Cabling to the sub-board in the battery room** 
The available charge rate for the battery will be limited.  This figure is more dictated by the underground cables 16mm2 about 96 Amps (in AU) but if the car is charging it is limited to 7.2 Kw.  If the battery was charging at the same time assume 7.2Kw the product would be a 64 Amp total load.  Acceptable. 

The last power hungry thing is the air-conditioner.  Yes it is possible and not a good thing, to have a battery charging and a car charging with an air-conditioner running  ie 22.5 Kw or 96 Amps or there abouts.  This may pop the main breaker rated at 80 Amps but whilst I think this is an unusual occurrence, there are protections although I not immediately thinking that will happen as the air-conditioner is an inverter variety it uses about 2 Kw/hrs once established.  therefore I'm not expecting any supply interruption not, that I'm setting out to do it or keen to test it.

**Other Cable install (wiring needs)**
Will isolate the battery from the main building.  I have a well ventilated space in the workshop for the battery inverter, the control gear and the associated sub switchboard equipment  
I'm currently in the ground installing cables.  At the main house switchboard, I have installed a new main switch 80 Amp a 32 amp breaker to pick up the car charger, a new 63 Amp breaker as well as a programable timer which also has a built-in 63 Amp breaker.  The outlet of the 63 Amp circuit goes underground to the new sub-switchboard. 

The sub-switchboard has a double isolator, switching both conductors at 63 Amps. 
Within this switch board there is enough room for to 2 LillyGo boards a circuit breaker for power to these boards, two other circuit breakers for the EV battery contactors.  I'm confident about the process to of building the power supplies but what do I use for the power supply since the grid power will be turned off for 6 hours every day.  Should I have a 12 volt lion battery maintained by the mains power or should this be maintained similar to the Model 3 accessories battery.  I guess I need help advice for a staring place. 

**Help to determine the health of an EV battery removed from a wreck.** 
Trying to source an EV battery in Australia is a bit pricy.  62Kw batteries sell for around $8-10K if you are lucky you may find them a bit cheaper but knowing the post accident condition and what has happened since the crash ie removal from car by the gorilla workforce  makes the spend risky.   Is there a way to test these batteries remembering the fuse may be blown and the likelihood of a refund from vehicle wreckers is not great.  Can you read live BMS data if a fuse has blown? 

Thanks in advance

josiahhiggs, 2024-12-26:
Good plan so far and what you are setting up is a grid tied energy storage system. Most inverters are able to operate without PV input and can be manually controlled or use a home assistant type of program to automate the charge/discharge times. You are able to check used EV batteries with the correct wiring setup and programs. I think you would have all your questions answered by reading all the discussions on the Discord/Patreon supported site. [https://www.patreon.com/dala](https://www.patreon.com/dala) We have channels for each battery and inverter that is working and ones in R&D. You will also find a good number of auzzie mates doing just what you are wanting to do!

Dutchlincoln, 2024-12-26:
Hi, where can i find the discord?

  

Thank you.

  

Kind regards,

  

Stef.

 �

josiahhiggs, 2024-12-26:
@Dutchlincoln follow the link to Dala's Patreon page and click on the membership tab, become a supporter to gain access to the Discord. [https://www.patreon.com/dala](url)

KenG-TeslaAU, 2024-12-27:
Thank you very helpful.  Lots of reading to do.
Groperfish

mvgalen, 2025-01-02:
In short: Yes this is possible, but further questions are probably better taken to the discord channels.
However, even with trees around, I would be amazed if some Solar panels especially with a battery setup would not be feasible.

KenG-TeslaAU, 2025-01-03:
Hi,
Thanks for responding and the direction to the discord channels.
  
As for panels You are completely right, even low sun can provide some useful energy generation. 

You can waste some time here but feel free not to bother but here are some reasons I chose batteries above panels. 

We did think through panels extensively but the lower efficiency coupled with fire risks, wild life nesting, falling branches, leaf litter, bark sheading, eucalyptus mist and seed pods all conspire to get trapped or cover the panels on this 18 degree roof pitch.  

Fire risk - My area High 
All of the nature wild life and natures waste products enhances the actions of an ember storm providing more than enough fuel to completely destroy my home.  I'm not lazy, we blow the roof off 3 times a week in late spring through to early autumn. The grounds require raking just about every day to keep fuel loads lower. Any panel array would need to be cleaned monthly due to sap mist from the eucalypts that forms a translucent film on the panel. The same sap falls on our glass skylights.  Its a bugger to get off.  I'm OK to do this cleaning now but I'm getting on, and,  can I do this when I'm 80? 

Costings 
We have a smart meter.  I regret this. 

When you do the sums a 12 to 14 Kw array will cost around $18K in Australia.  As you'll need to factor in micro inverters and leaf guard and bird netting and panel lifts.  Lets say a good day 6 KW/hour say 5 hours total 30KW.  Cant use 30Kw unless the car needs a charge once a week. The cost to sell this is about 4.9c to 6.3c  ie roughly $1.80 a day.   Between the hours of 9:00am to 4:00 pm.  We don't need the power then.  We need it later to cook and maybe run a air conditioner.  

During price spikes power can cost $20.20/Kw, Recently there were 13 instances of these in December 2024 some went from 5:00 pm to 9:00 pm. Even switching every possible device OFF we were still running a refrigerator 250W/hrs, internet/landline and cable box 36W/hrs,  1 light 16W/hrs, a fountain pump 32W/hrs, computer laptops with separate screens x 2 and a TV  75W/hrs.  This modest usage totals 400W/hrs giving a cost during a spike of just 1.6Kw/hrs and a cost of $32.32 x 13 = $420 for the month. Note we are not using an oven or air conditioning. 

In addition, if you are still with me,  you will love this, the Owner of the Poles and Wires charge a daily access fee. This is separate to my usage supplier company.  The poles and wires company for my area is AusGrid their fee is usually about 80 cents a day.  But they also charge a demand fee.  

The demand fee is for 8 months of every year summer and winter and the shoulder period either side, between the hours of 3:00pm to 9:00pm. every day.  They look for the largest watts consumed in any half hour block within the demand fee window.
They multiply this by 2 to get Kw/hrs 
They multiply the answer by the number of days in the month 
Then they multiply that answer by their Kw/hr fee and add it to the bill usually around 38 cents per Kw/hr  

Lets plug in some values.  Say I use my 2400W sandwich press once in the month for 15 minutes.
The half hour block will record 600W but my frugal usage is another 400W.  the total for the half hour is 1000W  They will double this to 2000W to get 2Kw
They multiply this by the days in the month say 30 days by 2Kw = 60Kw/hrs  
They multiply this by 38 cents answer $22.80. 

So whilst I would like solar my spend needs to be considered, without a battery solar gives me little advantage. 
A battery will give me a good advantage as I can avoid price spikes and demand period costs 
A battery and solar is my best option but the cost is more than a battery project 

I apologize for the length of this but I did not to appear to be dismissive of your comments.  Clearly you have much more knowledge than me and I appreciate your thoughts and time.  

Today I be began installing the remote wiring.  The next part of this project, a Tesla M3 battery 78Kw or 2 commercial Chinese 30Kw batteries with built in inverters. Both are around the same price of $12K.  Both a fraught with difficulties....maybe I like hard.

Cheers

Ken