dalathegreat/Battery-Emulator #694: Error while burning initial code.

windydrew, 2024-12-22:
### Expected Behavior

Followed directions to upload latest code.

### Actual Behavior

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4916
load:0x40078000,len:16436
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3524
entry 0x400805b8
__ OK __
EEPROM was initialized for event logging
head: 0, tail: 0
No events in log
Warning: The board was reset due to an exception or panic. Inform developers!
Creating Access Point: Battery Emulator
With password: 123456789
Access Point created.
IP address: 192.168.4.1
Connecting to Wi-Fi...
assert failed: tcp_alloc /IDF/components/lwip/lwip/src/core/tcp.c:1851 (Required to lock TCPIP core functionality!)

Backtrace: 0x40082649:0x3ffb3010 0x4008e4c9:0x3ffb3030 0x400948be:0x3ffb3050 0x40101b73:0x3ffb3180 0x40101ced:0x3ffb31a0 0x400dd318:0x3ffb31c0 0x400e308a:0x3ffb3210 0x400d90b9:0x3ffb3230 0x400d3332:0x3ffb3280 0x4008f212:0x3ffb32a0

ELF file SHA256: 69900b1c5

### Settings

- Software version: `7.0`
- Battery used: `Test`
- Inverter communication protocol: ``
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `no`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2024-12-22:
Thanks for the report. I added a note about this on the release page for 7.9.2 that ESP32 v3.0.7 or lower is required.

[image]

We will release 8.0.0 shortly, to support the latest ESP32 v3.1.0 release!

josiahhiggs, 2025-01-01:
Looks like PR730 will also fix this issue, @windydrew  would be great to check this setup now.

dalathegreat, 2025-01-02:
Closing this issue as fixed, v8.0.0 has now been released with support for the ESP32 v3.1.0 :partying_face:

closed 2025-01-02 by dalathegreat