dalathegreat/Battery-Emulator #696: Lilygo GPIO pins with SMA SBS inverter

tjjq44, 2024-12-23:
Hi, I plan tu buy a SMA Sunny Boy Storage 3.7 to reuse my Peugeot Ion 16kWh battery. According to the wiki, I need at least : 
- isolated can channel (I plan to use MCP2515)
- automated contactor control
- additional enable line

The latter uses GPIO5, but it is already assigned to MCP2515 SDI input... Is there any way to work around this? Maybe there is undocumented but usable IO pins elsewhere on the board?

Before I go any further with my project and invest in an inverter I really need to sort out this issue.

dalathegreat, 2024-12-23:
Hi @tjjq44 , we are testing a new solution for this, adding a CAN filter. This way does not consume any extra IO pins on the board, and you get the isolated CAN as required by the inverter. See this page for more info: https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-filter-hardware

lenvm, 2025-01-02:
Hi @tjjq44, please checkout the [ESP32 DevKit V1](https://github.com/dalathegreat/Battery-Emulator/wiki/Hardware:-ESP32-DevKit-V1) hardware. This hardware has enough pins to support your setup.

dalathegreat, 2025-01-03:
And a third (cheapest) option is to use a SD card breakout board, to get another free pin on the LilyGo 

https://shop.exploitee.rs/shop/p/microsdemmc-breakout-board

Anyways, tons of options! 

Fourth option is to go with the Stark hardware, it has extra pins available too!

I'll close this issue as completed!

closed 2025-01-03 by dalathegreat