dalathegreat/Battery-Emulator #701: ATTO 3 Battery SOC Discharge Behavior Issue

YAQ0T, 2024-12-25:
# ATTO 3 Battery SOC Issue Report  

## Overview  
This document describes an issue encountered with the ATTO 3 battery and inverter system, where the State of Charge (SOC) does not behave as expected during discharging.  

---

## Expected Behavior  
When the ATTO 3 battery is fully charged:  
- The inverter should display a 100% SOC.  
- As the battery is discharged, the SOC should gradually decrease until it reaches the programmed cutoff point.  

---

## Actual Behavior  
The SOC behavior is inconsistent and displays abrupt drops:  
- After a full charge, the SOC drops directly from 100% to 70% and then to 50%, while the voltage remains above 420V.  
- As the voltage decreases to 418–419V, the SOC suddenly drops further to 19%.  
- The SOC reaches 0% when the voltage falls to approximately 405V.  

### Voltage Configuration  
- **Maximum voltage:** 430V  
- **Minimum voltage:** 370V  

---

## Steps to Reproduce  
1. Charge the ATTO 3 battery to full (430V).  
2. Start using the inverter and monitor the SOC behavior as the battery discharges.  
3. Observe the SOC readings and corresponding voltage levels throughout the process.  

---

## Settings  

| Parameter                     | Value   |
|-------------------------------|---------|
| **Software version**          | `7.9.2` |
| **Battery used**              | `ATTO 3` |
| **Inverter communication protocol** | `CAN` |
| **Hardware used for Battery-Emulator** | `LILYGO` |
| **CONTACTOR_CONTROL**         | `yes/no` |
| **CAN_ADDON**                 | `yes/no` |
| **WEBSERVER**                 | `yes/no` |
| **MQTT**                      | `yes/no` |

---

## Additional Information  
Below are some images illustrating the observed behavior:  

1. [image]  
2. [image]  
3. [image]  
4. [image]  
5. [image]  
6. [image]  

---

YAQ0T, 2025-01-04:
Hi guys any updates ?

dalathegreat, 2025-01-04:
Really not possible to make the estimated SOC% based on voltage better. The LFP has such a flat charging curve, so any load from the inverter will significantly impact the reading.

We will try to add crash-unlocking to the software as soon as someone can provide log files: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BYD-Atto-3#how-do-i-unlock-a-crashed-battery

GraphicObession, 2025-02-04:
Hi Dala 

I happy to help with providing log files, however the link above takes me to a blank page. 
Do you need CAN log files from the Battery Emulator?  In what states should the devices be in to capture the logs?

YAQ0T, 2025-03-10:
Hi @GraphicObession , The link works perfectly for me .

lenvm, 2025-03-27:
This should be the latest working link:
[BYD Atto 3 - how do I unlock a crashed battery](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BYD-Atto-3-%E2%80%90-Yuan-Plus#how-do-i-unlock-a-crashed-battery)

dalathegreat, 2025-05-05:
Crash unlocking has now been merged to main (8.12.dev), and Wiki has been updated with info on how to use it: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-BYD-Atto-3-%E2%80%90-Yuan-Plus#how-do-i-unlock-a-crashed-battery

If this does not work, let's open a new issue! Closing this!

closed 2025-05-05 by dalathegreat