dalathegreat/Battery-Emulator #729: ESP32 Board 3.1.0 Event/Errors after update

josiahhiggs, 2025-01-01:
### Expected Behavior

Update board manager to ESP32 3.1.0 
Existing hardware, Lilygo + MCP2515 or MCP2518
Compile with Main branch 8.0.dev
Resume operation of system using latest board manager and 8.0.dev

### Actual Behavior

Lilygo and MCP2515 or MCP2518 is powered as directed in [Wiki](https://github.com/dalathegreat/Battery-Emulator/wiki/CAN-add%E2%80%90on-(MCP2515)#option-1-mcp2515-powered-via-5v-easiest-no-solder-method)
ATO webpage is stuck waiting for webserver. LED turns on solid red. No response from Lilygo when pressing reset or boot.

### Steps to Reproduce the Problem

1. Update Board manager ESP32 to 3.1.0
2. Compile and upload via USB C without additional power supply to Lilygo, connected to battery and inverter CAN.
3. Events Logged in Webserver 
[image]
4. Remove USB C and reconnect Lilygo to dcdc power supply of 5v at 2 pin connector, no LED on Lilygo, MCP showing power LED on.

### Settings

[USER_SETTINGS.h.txt](https://github.com/user-attachments/files/18286443/USER_SETTINGS.h.txt)

- Software version: 8.0.dev 
- Battery used: Tesla M3 LFP 
- Inverter communication protocol: SolaX CAN 
- Hardware used for Battery-Emulator: HW_LILYGO, MCP2515 to Inverter
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `yes`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2025-01-01:
Thanks for the bug report. The USER_SETTINGS helped to reproduce the setup required to troubleshoot this.

Issue seems to be an assert failure at boot; `assert failed: prvInitialiseNewTask tasks.c:1088 (uxPriority < ( 25 ))` 

Still investigating how to fix the issue :mag:

dalathegreat, 2025-01-01:
Fix has been deployed to main! :partying_face:

closed 2025-01-01 by dalathegreat