dalathegreat/Battery-Emulator #736: The new 8.0.0 release with esp 3.1 is overloading the CAN bus for Solax Ultra

JorgenSeemann, 2025-01-02:
### Expected Behavior

Normal behaviour

### Actual Behavior

The CAN / Lilygo is pushed so hard the webserver gets unresponsive.
CANFD buffer full is repeaded all the time

1.
1.
1.

### Settings

- Software version:  8.0.0
- Battery used: `` KIA E-GMP 77kwh
- Inverter communication protocol: Solax
- Hardware used for Battery-Emulator: `HW_LILYGO, + FD board
- CONTACTOR_CONTROL: no
- CAN_ADDON: yes
- WEBSERVER: yes
- MQTT: no

JorgenSeemann, 2025-01-02:
user settings
[USER_SETTINGS.zip](https://github.com/user-attachments/files/18292327/USER_SETTINGS.zip)

dalathegreat, 2025-01-07:
This PR should help: https://github.com/dalathegreat/Battery-Emulator/pull/756

dalathegreat, 2025-01-13:
@JorgenSeemann , could you try the latest main (8.2.dev)

We have both CAN-FD fixes, and a new lightweight Webserver library. Both these should help your issue!

JorgenSeemann, 2025-01-13:
Seems better already with 8.1 :)

dalathegreat, 2025-01-20:
Closing as fixed, 8.2.0 further improves performance for FD! 👍 Thanks for reporting and testing!

closed 2025-01-20 by dalathegreat