dalathegreat/Battery-Emulator #737: Question: build battery emulator for PC or SBC

kyberias, 2025-01-03:
This can be labeled as a question.

Has there been some efforts or discussions to build Battery emulator for a PC (Linux or Windows) environment? 

Currently the code has quite a lot of direct dependencies to Arduino/ESP libraries etc. but looks like the core functionality and "business logic" could probably be built and run in a proper OS and with standard PC serial ports etc. This includes SBCs such as Raspberry PI, or course. GPIO might be something of a challenge in a normal PC. This build might be accomplished with a couple of pre-processor defines and of course by implementing comm-code for PC environment.

I for one might prefer to run the code on more robust environment with Ethernet etc. rather than ESP32.

dalathegreat, 2025-01-03:
Raspberry PI variant is available here: https://gitlab.com/pelle8/batteryemulator

closed 2025-01-17 by dalathegreat

nagyrobi, 2025-01-17:
Any PC based approach is by far worse in aspects of robustness and reliability than an MCU solution. There are ESP MCU boards with Ethernet if WiFi is a concern.

kyberias, 2025-01-19:
@dalathegreat  Regardless of other repos, would this project accept PRs with the intention of building the C/C++ code for PC platform?

dalathegreat, 2025-01-19:
Since PCs are generally not considered real-time systems, it is best to fork this and implement a new project that just syncs with the battery/inverter protocol definitions!

kyberias, 2025-01-19:
Right, ok. I'm curious about the real-time requirements though. What parts of the battery emulator would require real-time?

dalathegreat, 2025-01-19:
@kyberias The CAN sending routines run on fixed 10ms loops. If the system is not able to transmit periodically (2.5ms deviation allowed), many batteries will pick up on this and throw errors. That is why porting this to a Windows machine is a bad idea.

kyberias, 2025-01-19:
Got it, I didn't quite realize battery comm can be so sensitive. But I wonder how Per Carlen's Python code is able to guarantee such requirements running on Raspberry Pi. Pi Linux or Python most certainly can't guarantee any millisecond intervals any more than normal Windows can.

dalathegreat, 2025-01-19:
The current RPI code only supports LEAF and Tesla batteries, which are not sensitive at all when it comes to periodicness of messages. But many packs like BMW, MEB platform, eGMP, +more are extremely sensitive.