dalathegreat/Battery-Emulator #740: BMW i3 Power value displays 0 for 2nd battery when SoC scaling turned off

pmajor021, 2025-01-03:
### Expected Behavior

it should display a value

### Actual Behavior

stays on 0 all the time if SoC rescaling is off

### Steps to Reproduce the Problem

open the webpage

### Settings

- Software version: `8.0.0`
- Battery used: `BMW i3`
- Inverter communication protocol: `BYD ModBus`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `yes`
- WEBSERVER: `yes`
- MQTT: `yes`

Found the issue in software.ino: the power calculation for battery2 is within the part of code when the SoC rescaling is active.

I put that calculation in front of that 'if':

```
  /* Calculate active power based on voltage and current*/
  datalayer.battery.status.active_power_W =
      (datalayer.battery.status.current_dA * (datalayer.battery.status.voltage_dV / 100));

  #ifdef DOUBLE_BATTERY
    /* Calculate active power based on voltage and current*/
    datalayer.battery2.status.active_power_W =
        (datalayer.battery2.status.current_dA * (datalayer.battery2.status.voltage_dV / 100));
  #endif  // DOUBLE_BATTERY

  if (datalayer.battery.settings.soc_scaling_active) {
    /** SOC Scaling
```

pmajor021, 2025-01-03:
```
@ -350
  datalayer.battery.status.active_power_W =
      (datalayer.battery.status.current_dA * (datalayer.battery.status.voltage_dV / 100));

+  #ifdef DOUBLE_BATTERY
+    /* Calculate active power based on voltage and current*/
+    datalayer.battery2.status.active_power_W =
+      (datalayer.battery2.status.current_dA * (datalayer.battery2.status.voltage_dV / 100));
+  #endif  // DOUBLE_BATTERY

  if (datalayer.battery.settings.soc_scaling_active) {
    /** SOC Scaling
     * 

@401
#ifdef DOUBLE_BATTERY
-    /* Calculate active power based on voltage and current*/
-    datalayer.battery2.status.active_power_W =
-        (datalayer.battery2.status.current_dA * (datalayer.battery2.status.voltage_dV / 100));
-
    // Calculate the scaled remaining capacity in Wh
    if (datalayer.battery2.info.total_capacity_Wh > 0 && datalayer.battery2.status.real_soc > 0) {
      calc_max_capacity =

```

dalathegreat, 2025-01-12:
Thanks for the report. Fixed in this PR https://github.com/dalathegreat/Battery-Emulator/pull/786

closed 2025-01-12 by dalathegreat