dalathegreat/Battery-Emulator #755: CHAdeMO Support Status

yngndrw, 2025-01-06:
Hello,

I'm wondering what the status of the CHAdeMO support is, as the page mentions that it's for emergency use only right now. Aside from the mention of contactor welding in other V2X setups and the cable sourcing issues, what are the current limitations of this setup? We're interested in this setup with a Nissan e-NV200, as that is the only VAN that we know of with V2X support. Given the status of V2X right now, we feel that a roll-your-own setup is the only viable option. (We need a van anyway, and it's easier to have a dual-use battery in the driveway than having to find space and purchase an extra battery pack. An EV van has other benefits for camper use, as well)

There seems to be very little detail on the page, especially around the shunt usage. Does anybody have a proper wiring diagram?

Thank you.

Ginger-blackman, 2025-01-12:
I was wondering about this as well. I've got a 2015 Leaf that I'd like to use as a house backup since it mostly just sits in the driveway while we use our '22 Leaf to get around.
Currently we have a EG4 (Luxpower) 18kPV inverter with an EG4 powerpro battery. I was looking at either connecting the '15 Leaf through Chademo to the EG4 inverter through a DC-DC converter to match the 48v system, or get a Fronius inverter to use just for converting power from the HV battery.

I'd love to help more with the battery-emulator project! More than happy to get logs from the EG4 equipment if needed.

yngndrw, 2025-01-14:
Looking through the code, there's mentions of various pages within the spec - Does anybody have a link to a copy of the protocol specifications? I couldn't find more than an overview myself.

yngndrw, 2025-01-17:
FYI I found this Chademo cable for a good price, although sadly it says it's not available in my region (UK) - It does however show up on the US site for anyone over there:
https://www.aliexpress.us/item/3256807949751424.html

mvgalen, 2025-01-25:
Just mind that the initial shown price is without the cable (357 euro), with cable it is 235 euro higher.

HansNetcom, 2025-03-13:
Hi , I also have a 2024 Nissan Leaf for my wife that stand most of the time in our garage.
It would be perfect home-battery if connected to my Goodwe GW6.5K-ET hybrid via the CHAdeMO plug.
However, I did not see a save and tested installation yet. 
Is there already anyone that has this option running and using it comfortably?
I,m very interested to hear

HansNetcom, 2025-04-23:
Hi.
At the moment I have a Fronius Symo Hybrid 5.0 3-s installed  for use with my Nissan Leaf 2023 as home-battery. 
The battery-emulator is working or the inverter. 
I,m using a CHAdeMO plug for the connection to the Nissan Leaf.
At the moment I,m struggling with the connection of the CHAdeMO plug to the LilyGo board.
I can not find a complete connection diagram and I,m not sure if the code in the CHADEMO-BATTERY.cpp is based upon the  automatic pre-charge of the Nissan. In all the info I have read I could not find a complete  connection scheme. 
Also the Start/Stop implementation is not clear or me. Please help me to make the last steps in my project.

PeterOuwehand, 2025-04-29:
we have also an nissan E-NV200 here (the Netherlands), waiting for an release of the code. i also try to find an Chademo connector. but i give up and want to install an CCS2 connector on the car and use that connector to connect at home https://voldt.nl/collections/dc-snelladen 
we are still trying to get the Lilygo connected to the CAN lines of the Nissan. the schematics are on the internet (pin 8-9 for the CAN), we meassure 126 Ohm but no voltage difference. with the pin 1 and 7 connected we hear an relais click but still no voltage difference on the CAN. does somebody already have the connection running?

HansNetcom, 2025-04-30:
At the moment I have the canbus connection running and was able to read all incoming and outgoing messages. My Nissans was not correct responding on the messages in the software.
I had to re-engineer the complete canbus message sequence to make my Nissan Leaf giving the correct answers. ChatGPT is a great help in this.
For the Chademo connection I used the 3D printed version. It’s save for inside and dry environment and I use it for testing only.
The connection of the Chademo connection to the Lilygo  is again challenging but I,m making the right steps. Unfortunately I made a wiring mistake and bluwup the Lilygo. Waiting for AliExpress for a new one.
To test the wiring I made a simple testprogram to check the correct wiring.
If this works as expected I consider to buy a professional Chademo plug and kabel for about 350€
To be continued.

PeterOuwehand, 2025-05-01:
Can you share how you make connection to the car? do you put 12V to the car for starting the communication?
and where did you find an supplier for the cademo plug with cable for €350,-?

HansNetcom, 2025-05-01:
If everything is working as it should I will share my complete experiences.

HansNetcom, 2025-05-23:
**Update**
Still struggling to connect my leaf to the BE.
Can-bus communication works fine,even with the shunt installed, however I get  the wrong SoC from my leaf, real value is 68% , in the web-server and via MQTT it reports 0.68%. 
So I get no contactor closing from the leaf.
btw, With the FAKE-BATTERY it looks fine.  ??

Also the MQTT communication between my HA and the BE needs some attention to instruct the BE to charge or discharge.
Finally this should be addressed automatically based on the tariffs of my Dynamic Energy Contract, Yield from the solar panels, internal energy usage with the ultimate goal 0 on the energy meter. ;-)

**CHADEMO connector and cable.**
I got an offer from China (www.chinaevse.com) for a  CHAdeMO connector including 8M kabel for 80A. See attached PDF
If my project is working properly I will replace my 3D printed CHAdeMO plug with this one. 
If some one in the Netherlands is interested to buy a couple of CHAdeMO plus cable we could shared the delivery costs. 
	
[PI-HW20250322M.pdf](https://github.com/user-attachments/files/20410122/PI-HW20250322M.pdf)