dalathegreat/Battery-Emulator #757: In-vehicle Inverter

yngndrw, 2025-01-06:
This is a little off-topic for this project, but I wanted to ask here because there's a lot of knowledge and experience here around solar hybrid inverters.

Long story short, we are planning on a Nissan e-NV200 camper conversion and would like a decent 230V AC supply without having to muck around with dual conversion and extended leisure batteries. I know a lot of people do that, but it has always seemed like a weird setup to me. Nissan did provide a 1,500W AC outlet in the Jap models, but it wasn't offered elsewhere. The idea is to draw power from the HV system for the inverter and to supplement this using solar panels. Any charging would probably be done via the built-in vehicle charger for simplicity regarding the BMS.

I have a few questions about the way a non-grid-tie hybrid inverter would work:
1) Usually the solar input is limited by the solar panels - But if it was connected to an unrestricted source, am I right in presuming that a typical inverter would **not** limit the draw and would become a magic smoke machine? Or would it self-limit to the maximum rating of the inverter?
2) Using the battery input, does the BMS communication need to be bidirectional if it's only drawing from the battery and is not charging it? The reason I ask is because I envision the need for a CAN gateway which allows the hybrid inverter to eavesdrop on the BMS (So that it can't over-discharge it), but does not allow the inverter to interrupt the vehicle's EV CAN bus.
3) Can a typical hybrid inverter be set to not charge the battery and to only discharge it?

Thank you for any help.

beadon, 2025-01-07:
1. The inverter has limits which would would set, get an inverter that has the ability to se these limits so too much or too little power is not a problem.  That being said .. I am not even sure what inverters would be fit for purpose, I have mostly been using Solis.

2. You need connectivity to the BMS to tell the battery system what to do - there are internal contactors and a lot of messages that tell the BMS to function.  Without it, the battery will not supply power to the contactors.

3. Yes, but each model is different.  Unfortunately I do not have a lot of inverters to choose from, definitely one for deeper research.

yngndrw, 2025-01-07:
Thank you.

I did see some videos from Benjamin Nelson directly connecting a solar inverter to CHAdeMO (Via the PV inputs, not as a battery) without a BMS connection - It seemed to work in his test but in that video he raised some concerns about the MPPT system shorting out the battery so that's why I wanted to check. It's good to know that some inverters can be configured with limits as that would resolve the issue so I'll have a look to see which types can do that.

I'm not too worried about the internal contactors as I can just switch the vehicle on - I'd need to do this to get the internal HVAC system working anyway. I think over-discharge is my biggest concern.

It sounds like it's all possible though, I just need to read through a load of inverter manuals to make a shortlist. Thanks again.

beadon, 2025-01-07:
When you are running the car attached to the battery AND you are drawing power from it separately, I expect that you will trip a fault and the BMS will shut off the battery.  This is a safety measure to make sure that power is not unexpectedly getting depleted (causing a fire perhaps?).  I remember reading this on a Tesla forum somewhere, so I expect it to be common safety practice that drawing more than a certain amount of power while still attached to the car will trigger some kind of shutdown.  Of course, this is likely to differ between manufacturers, and I would watch this closely during testing.

yngndrw, 2025-01-07:
Ah that's interesting and good to know, I hadn't heard of that. I had just assumed that once the vehicle was on, the HV circuit was live and anything could tap into it - Like the DC-DC converter, PTC heater, air conditioning compressor and the AC inverter (In some regions) currently do. Hopefully, the Nissan platform doesn't have that feature but I'll think of a way to test it before spending too much on the project. I have seen that some people have installed a second charger in parallel in the Nissan Leaf (To increase the charge rate from 3.3kW to 6.6kW) so I'm hoping that as that worked fine, discharging also isn't monitored.