dalathegreat/Battery-Emulator #761: Migrate MQTT library to newer supported version

dalathegreat, 2025-01-07:
The currently used MQTT library stopped getting updates in 2020. This is the one currently in use: https://github.com/knolleary/pubsubclient

It would be better from a cybersecurity perspective to migrate to one that is getting updates, especially now that we are starting to introduce MQTT commands such as BMS reset and SOC recalibration.

Options:
A: Easy quickfix, This fork seems more maintained: https://github.com/thingsboard/pubsubclient
B: More effort, espressif mqtt library https://github.com/espressif/esp-mqtt

StarkJohan, 2025-01-07:
I agree that leaving pubsubclient is a good thing and I don't think a fork of it is a good solution either. It's not supporting all MQTT functions. Another option is "arduino-mqtt" which has a much better support for MQTT functions than "pubsubclient" has:

https://github.com/256dpi/arduino-mqtt

However, from what I can tell the espressif library also seem to cover the functions that pubsubclient is missing and thus the espressif option is probably the smarter choice in the long run.

dalathegreat, 2025-01-20:
Closing as completed, this is now in main and next release this will be included in is 8.3.0 🥳

closed 2025-01-20 by dalathegreat