dalathegreat/Battery-Emulator #767: Solis Inverter did not stop discharging battery when minimum voltage was reached

NotSoProTip, 2025-01-07:
### Expected Behavior

Decrease amps until 0 so pack voltage does not dip below set limit. (As it does when max voltage is reached during charge)

### Actual Behavior

Continues discharging the battery way past set limit.

### Settings

- Software version: 8.1
- Battery used: Test_Fake_Battery
- Inverter communication protocol: Pylon_CAN
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `no`
- WEBSERVER: `yes`
- MQTT: `no`

dalathegreat, 2025-01-10:
Massive thanks for the bug report @NotSoProTip , I have implemented a fix for this issue here: https://github.com/dalathegreat/Battery-Emulator/pull/778

dalathegreat, 2025-01-12:
Closing this as done. Not possible to fix with Pylon protocol, inverter seems to still consume power via selfconsumption.

For future implementation, we can look into wht Solis inverter would need to use BYD_CAN

closed 2025-01-12 by dalathegreat