dalathegreat/Battery-Emulator #775: Tesla, not possible to compile DOUBLE_BATTERY with main

dalathegreat, 2025-01-10:
### Expected Behavior

DOUBLE_BATTERY and TESLA_MODEL_3Y_BATTERY compiles

### Actual Behavior

Compilation errors

```
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp: In function 'void handle_incoming_can_frame_battery2(CAN_frame)':
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp:1968:7: error: 'battery2_regenerative_limit' was not declared in this scope
 1968 |       battery2_regenerative_limit =
      |       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp:1970:7: error: 'battery2_discharge_limit' was not declared in this scope
 1970 |       battery2_discharge_limit =
      |       ^~~~~~~~~~~~~~~~~~~~~~~~
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp:1972:7: error: 'battery2_max_heat_park' was not declared in this scope; did you mean 'battery2_max_temp'?
 1972 |       battery2_max_heat_park =
      |       ^~~~~~~~~~~~~~~~~~~~~~
      |       battery2_max_temp
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp:1974:7: error: 'battery2_hvac_max_power' was not declared in this scope
 1974 |       battery2_hvac_max_power =
      |       ^~~~~~~~~~~~~~~~~~~~~~~
/home/dala/Git/Battery-Emulator/Software/src/battery/TESLA-BATTERY.cpp:2198:9: error: 'HVP2_gpioHvpSelfEnable' was not declared in this scope; did you mean 'HVP_gpioHvpSelfEnable'?
 2198 |         HVP2_gpioHvpSelfEnable = ((rx_frame.data.u8[1] >> 1) & (0x01U));         //: 9|1@1+ (1,0) [0|1] ""  Receiver
      |         ^~~~~~~~~~~~~~~~~~~~~~
      |         HVP_gpioHvpSelfEnable
```

### Steps to Reproduce the Problem

1.DOUBLE_BATTERY
1.TESLA_MODEL_3Y_BATTERY
1.

### Settings

- Software version: `8.1.dev`
- Battery used: `TESLA_MODEL_3Y_BATTERY`
- Inverter communication protocol: `none`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `no`
- WEBSERVER: `no`
- MQTT: `no`

josiahhiggs, 2025-01-15:
@dalathegreat, I can have a look into this one next.

lenvm, 2025-01-27:
Closing this as it is fixed in #769

closed 2025-01-27 by lenvm