dalathegreat/Battery-Emulator #785: MQTT_MANUAL_TOPIC_OBJECT_NAME seems to be mandatory since 8.1.0

pmajor021, 2025-01-12:
### Expected Behavior

until now I used MQTT settings on the way MQTT_MANUAL_TOPIC_OBJECT_NAME commented out

### Actual Behavior

now with 8.1.0 it gives a compilation error
/sketch/src/devboard/mqtt/mqtt.cpp.o:(.literal._ZL14publish_eventsv+0x54): undefined reference to `ha_device_id'
collect2: error: ld returned 1 exit status
### Steps to Reproduce the Problem

1. enable MQTT
2. disable manual topic on user settings
3. compile

### Settings

- Software version: `8.1.0`
- Battery used: `BMW_I3_BATTERY`
- Inverter communication protocol: `BYD_MODBUS`
- Hardware used for Battery-Emulator: `HW_LILYGO`
- CONTACTOR_CONTROL: `no`
- CAN_ADDON: `yes`
- WEBSERVER: `yes`
- MQTT: `yes`

dalathegreat, 2025-01-14:
@No-Signal , Wanna have a go at this one, or shall I?

No-Signal, 2025-01-14:
I have fixed this in my MQTT PR that I will push up shortly. Would you like the fix adding before as its a big PR? I think it was a 3 character fix.

dalathegreat, 2025-01-20:
This is now fixed in 8.3.dev (main). Target release is 8.3.0 👍

closed 2025-01-20 by dalathegreat