dalathegreat/Battery-Emulator #792: PAUSE_END phase trigger BATTERY_VALUE, BATTERY_OVERVOLTAGE and SOC_PLAUSABILITY_ERROR

luipir, 2025-01-14:
### Expected Behavior

Normal work of the emulator wthout triggering any alarm or error

### Actual Behavior

when PAUSE_END is reached, a set of error ar triggered as in the image
[image]

The error is triggered every day as 18:06 

### Steps to Reproduce the Problem

### Settings

detailed configuration:
https://github.com/luipir/Battery-Emulator/commit/71e3fdefa1db1c0ab3d30dd18d3fff8fa14c62d2

in resume the conf is:
```
#define NISSAN_LEAF_BATTERY

/* Select inverter communication protocol. See Wiki for which to use with your inverter: https://github.com/dalathegreat/BYD-#define BYD_CAN  //Enable this line to emulate a "BYD Battery-Box Premium HVS" over CAN Bus

/* Select hardware used for Battery-Emulator */
#define HW_STARK

/* Contactor settings. If you have a battery that does not activate contactors via CAN, configure this section */
#define PRECHARGE_TIME_MS 500  //Precharge time in milliseconds. Modify to suit your inverter (See wiki for more info)
#define CONTACTOR_CONTROL  //Enable this line to have the emulator handle automatic precharge/contactor+/contactor- closing sequence (See wiki for pins)
#define PWM_CONTACTOR_CONTROL  //Enable this line to use PWM for CONTACTOR_CONTROL, which lowers power consumption and heat generation. CONTACTOR_CONTROL must be enabled.
#define PERIODIC_BMS_RESET  //Enable to have the emulator powercycle the connected battery every 24hours via GPIO. Useful for some batteries like Nissan LEAF

#define CRYSTAL_FREQUENCY_MHZ 8  //CAN_ADDON option, what is your MCP2515 add-on boards crystal frequency?
#define CANFD_ADDON_CRYSTAL_FREQUENCY_MHZ \
  ACAN2517FDSettings::OSC_40MHz  //CANFD_ADDON option, what is your MCP2518 add-on boards crystal frequency?
#define WIFI
#define WIFICONFIG  //Enable this line to set a static IP address / gateway /subnet mask for the device. see USER_SETTINGS.cpp for the settings
#define WEBSERVER  //Enable this line to enable WiFi, and to run the webserver. See USER_SETTINGS.cpp for the Wifi settings.
#define WIFIAP  //When enabled, the emulator will broadcast its own access point Wifi. Can be used at the same time as a normal Wifi connection to a router.
#define MDNSRESPONDER  //Enable this line to enable MDNS, allows battery monitor te be found by .local address. Requires WEBSERVER to be enabled.
#define LOAD_SAVED_SETTINGS_ON_BOOT  // Enable this line to read settings stored via the webserver on boot (overrides Wifi credentials set here)

/* Battery settings */
// Predefined total energy capacity of the battery in Watt-hours
#define BATTERY_WH_MAX 30000
// Increases battery life. If true will rescale SOC between the configured min/max-percentage
#define BATTERY_USE_SCALED_SOC false
// 8000 = 80.0% , Max percentage the battery will charge to (Inverter gets 100% when reached)
#define BATTERY_MAXPERCENTAGE 8000
// 2000 = 20.0% , Min percentage the battery will discharge to (Inverter gets 0% when reached)
#define BATTERY_MINPERCENTAGE 2000
// 500 = 50.0 °C , Max temperature (Will produce a battery overheat event if above)
#define BATTERY_MAXTEMPERATURE 500
// -250 = -25.0 °C , Min temperature (Will produce a battery frozen event if below)
#define BATTERY_MINTEMPERATURE -250
// 300 = 30.0A , Max charge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_CHARGE_AMP 300
// 300 = 30.0A , Max discharge in Amp (Some inverters needs to be limited)
#define BATTERY_MAX_DISCHARGE_AMP 300
// Enable this to manually set voltage limits on how much battery can be discharged/charged. Normally not used.
#define BATTERY_USE_VOLTAGE_LIMITS true
// 5000 = 500.0V , Target charge voltage (Value can be tuned on the fly via webserver). Not used unless BATTERY_USE_VOLTAGE_LIMITS = true
// set to 4.1v per cell
#define BATTERY_MAX_CHARGE_VOLTAGE 3936
// 3000 = 300.0V, Target discharge voltage (Value can be tuned on the fly via webserver). Not used unless BATTERY_USE_VOLTAGE_LIMITS = true
#define BATTERY_MAX_DISCHARGE_VOLTAGE 3000
```

- Software version: `8.1`
- Battery used: `Nissan Leaf 30kWh Part number: 4NR4CTB`
- Inverter communication protocol: `BYD_CAN`
- Hardware used for Battery-Emulator: `HW_STARK`
- CONTACTOR_CONTROL: `yes`
- CAN_ADDON: `no`
- WEBSERVER: `yes`
- MQTT: `no`
- Inverter model: `Ingeteam Sun Play 6TL M`

dalathegreat, 2025-01-14:
Quick triage, The bug is most likely related to PERIODIC_BMS_RESET that is pausing the emulator every day.

closed 2025-01-20 by dalathegreat

luipir, 2025-01-23:
I'll give a feedback when new pause will be triggered automatically. Doing it manually do not trigger new error events (other that inverter that alert me because of lack of voltage reading). So IMHO  it could be considered solved

dalathegreat, 2025-01-23:
Super! BTW I have one more PR related to making the periodic pause better: https://github.com/dalathegreat/Battery-Emulator/pull/814

luipir, 2025-01-24:
FYI after moved to 8.2 the pause still trigger something (CAN_OVERRUN)

[image]

the inverter still alert me that a pause happen and lost BMS communication. speculating a possible configurable solution is to clone previous messages to avoid to alert some inverters during bms pause

dalathegreat, 2025-01-24:
Both those issues are what PR https://github.com/dalathegreat/Battery-Emulator/pull/814 is aiming to fix!