dalathegreat/Battery-Emulator #807: Read inverter data

eham365, 2025-01-20:
Hi. Could the Battery-Emulator support inverter information as well? For example use RS485 from the inverter to the RS485 on the CAN bridge and then MQTT? Many inverters are supplied with Solarman Dataloggers which has rubbish data update frequency.

dalathegreat, 2025-01-20:
At the moment there is no generic RS485 logger. All these RS485 implementations are inverter specific (Fronius has Modbus RTU via RS485, and Kostal has the propreitary serial ove RS485). This means we have no library for logging RS485 data easily.

I would be quite hesitant to add this, it sounds like it would be better to use a Raspberry Pi or similar to get the data logged faster than the manufacturer can provide. The Battery-Emulator hardware performs critical tasks, sending of messages and battery control, so just adding more loggers onto it is risky from performance side. That is why I would recommend going the RPi route here.

Closing this question as completed. For an example on how to get started with RS485 and RPi, checkout [Pelle_C's RPI system ](https://gitlab.com/pelle8/inverter_resources)

closed 2025-01-20 by dalathegreat

nagyrobi, 2025-01-20:
Plus, RS485 by design is something that you should know in advance. The slave device (eg. inverter) stays silent until asked by the master, but the master has to know in advance what to ask for. Shiffing this data is hard because slave iplementations can vary a lot between manufacturers. It's always better to get RS485 documentation from the manufacturer, it's explained clearly what answers give the slave to what queries. No need to log anything.