dalathegreat/Battery-Emulator #813: Cell Max and Min Values are not published via MQTT in Pylon battery mode, even visible on WebGui

cray9000, 2025-01-21:
By utilizing the Pylon Battery Protocol through CAN, the Battery Cell Voltage (maximum, minimum and delta values) are correctly displayed in the Web Interface. However, the individual cell voltages are not visible under the 'cellmonitor' section of the Web GUI in mycase.

Despite the correct values being available at least for the maximum and minimum voltage values, but they were not being propagated to Home Assistant via MQTT.

I have uncommented the 'if' clause in the mqtt.cpp file for my personal use:

      // publish only if cell voltages have been populated...
      //if (datalayer.battery.info.number_of_cells != 0u &&
      //    datalayer.battery.status.cell_voltages_mV[datalayer.battery.info.number_of_cells - 1] != 0u) {
        doc["cell_max_voltage"] = ((float)datalayer.battery.status.cell_max_voltage_mV) / 1000.0;
        doc["cell_min_voltage"] = ((float)datalayer.battery.status.cell_min_voltage_mV) / 1000.0;
      //}
	  
As a result, the MQTT values for the maximum and minimum cell voltages are now being propagated correctly.

It may be beneficial to further refine this 'if' clause specifically for Pylon mode. This would ensure that the values are propagated when the maximum and minimum voltages are available in the datalayer, even if `datalayer.battery.info.number_of_cells = 0u`, as the individual cell voltages are may not accessible in Pylon mode.

dalathegreat, 2025-01-22:
Ping @amarofarinha / @No-Signal , any input on this? We can either:

- Edit the MQTT code to not have this if statement.
- Edit the PYLON battery code to contain a cellvoltage value (2x cells, the max and min value)

dalathegreat, 2025-01-31:
Fix implemented in https://github.com/dalathegreat/Battery-Emulator/pull/841

lenvm, 2025-03-27:
As #841 was merged, I will close this issue.

closed 2025-03-27 by lenvm