dalathegreat/Battery-Emulator #822: Request for adding RJ45 Lan

PGcrypt, 2025-01-26:
Hi dala,
your project looks great :)
Do you plan to integrate support for an Lan RJ45 module like W5500?
There are already some librarys for that module, but my programming skills are far below yours :D

Thank you for that great repo!

dalathegreat, 2025-01-26:
@PGcrypt W5500 support can be added yes via GPIO. But may I ask, what is the usecase? Normally the Wireless Webserver is OK for everyone, and those concerned about cybersecurity run without any connectivity at all. So what does the LAN module solve?

PGcrypt, 2025-01-26:
I’m helping a friend build a DIY HV LiFePO4 system, and he turns off Wi-Fi at night (aluminum hat). So, I thought about using the eth.h library instead of wifi.h to address this.
Before I try implementing it myself, I wanted to ask if you’re planning to add such a feature or if you’d be interested in it if I contribute the implementation.
It might also be useful for others, as Wi-Fi often doesn’t work well in cellars.
Thanks for the fast response :)