dalathegreat/Battery-Emulator #824: RESET PANIC and Board Resetting -> MQTT?

obsoolete, 2025-01-27:
Running `8.0.0`, I experienced 3 "RESET PANIC" events over the course of about 1 week. Dala suggested MQTT was causing this problem, so I updated to `8.3-dev` (`a46c6f4a`), and after flashing the board became un-usable. I estimate every 30-60s the board would crash and reset. 

I enabled `DEBUG_VIA_USB` and was able to capture the log attached in **MQTTCrashLog.txt**. I have also uploaded my **USER_SETTINGS.h** file (the cpp is untouched, except for IP address). Of note in the **USER_SECRETS** file:

```cpp
#define WEBSERVER_AUTH_REQUIRED true
#define MQTT_USER ""
#define MQTT_PASSWORD ""
```

Am currently running 8.2.0 which has been solid for the last 30 hours.

N.B: I have home assistant connected to the MQTT server (with the MQTT integration), and previously there was nice parsing of all the messages from the battery emulator (i.e. recognised BE as a device and created entities for all the data). Since moving to 8.2.0, this has all disappeared. Messages are still being sent to the server (i.e. checked using MQTTX). Might have to make a separate issue for this.

[MQTTCrashLog.txt](https://github.com/user-attachments/files/18557676/MQTTCrashLog.txt)
[USER_SETTINGS.h](https://github.com/user-attachments/files/18557687/USER_SETTINGS.h.txt)

dalathegreat, 2025-01-27:
@obsoolete , the Wiki page for MQTT says;
"The username and password for MQTT can be set in the USER_SECRETS file. Incase you want to use no login info, set fields to NULL"

Does the crash happen if you set
```
#define MQTT_USER NULL
#define MQTT_PASSWORD NULL
```

dalathegreat, 2025-01-27:
Nevermind, I managed to get a crash reproduced with the settings file included. I dont have an MQTT server, so my crash might be different;

```
E (11085) esp-tls: [sock=49] select() timeout
E (11085) transport_base: Failed to open a new connection: 32774
E (11085) mqtt_client: Error transport connect
E (11249) outbox: outbox_enqueue(53): Memory exhausted
E (11251) outbox: outbox_enqueue(53): Memory exhausted
E (11253) outbox: outbox_enqueue(53): Memory exhausted
E (11255) outbox: outbox_enqueue(53): Me
abort() was called at PC 0x40182afb on core 0
```

Seems like the MQTT crashes the board if you try to connect to a server that does not exist. Think we need to fix this one also @No-Signal 😅

dalathegreat, 2025-02-02:
Closing this as completed, fix included in v8.4.0 🙌

closed 2025-02-02 by dalathegreat