dalathegreat/Battery-Emulator #825: 24kWh leaf wrong capacity, reset degradation is not working

Kontich, 2025-01-27:
Hello, thank you for great project <3 

I have question regarding capacity on webserver: it shows 21024Wh in settings, after changing this value to 24000 and refreshing page it goes back to 21024Wh. 
Battery Emulator 8.2.0
I am not sure if it is expected because my SOH is 57%

also I did try to reset degradation data but there is an error:

[image]

[image]

dalathegreat, 2025-01-27:
On the Nissan LEAF, the code uses automatic detection of kWh available. However, this only works for 30-40-62kWh batteries, we just assume that 24kWh are at 100% and have 21kWh net capacity (273GIDS). This could be improved by multiplying this value by SOH (*0.57). I will mark this issue as a bug, and we can work on it for the next release!

The Reset Degradation Data was mostly developed for 30kWh packs. There is no gain to reset capacity on 24kWh, but if you really want to do it, you can use the dedicated app developed by safetyuggs, it is available on the MyNissanLEAF forum!

dalathegreat, 2025-02-08:
This is now fixed with PR https://github.com/dalathegreat/Battery-Emulator/pull/862

closed 2025-02-08 by dalathegreat