dalathegreat/Battery-Emulator #828: Add in Fox-ESS proptocol as a battery option

simonjowett, 2025-01-28:
Given the Fox ESS protocol has been added to the list of inverters would it be possible to add this protocol as a battery option?  I notice this is now approved for use in Australia and has an advertised price of $4,949.81 Australian Dollars which is much less than the Tesla PW (about 1/3 the cost).  With the ubiquity of Fronius Gen24 inverters here in Oz it would be interesting to see an expansion of the very limited range of battery options.  I also read that the Fox ESS batteries have become popular in the UK also.

dalathegreat, 2025-01-29:
Thanks for the great suggestion @simonjowett , I have started development on this! Will let you know once it is ready!

dalathegreat, 2025-01-29:
Support for FoxESS batteries have been added in pull request https://github.com/dalathegreat/Battery-Emulator/pull/830

Ready for testing ✅

closed 2025-02-09 by dalathegreat