dalathegreat/Battery-Emulator #829: Better voltage estimaton for KIA 64,8kwh 37501-AO050

MrElCapitano, 2025-01-29:
This issue is related to #627 "better voltage estimation" due to the workaround that Vodynov did to be able to use the 37501-AO050 battery pack.

Hem mentioned that he have been using the battery with his workaround for over 2 month with no issues.

Is it possible for an update regarding better voltage estimation to support 37501-AO050 in EGMP or should it have a separate configuration choice?

/Capitan

ilia92, 2025-03-01:
I think this is related:
https://github.com/dalathegreat/Battery-Emulator/pull/923

lenvm, 2025-03-27:
I think this could be the relevant post from Vodynov: https://github.com/dalathegreat/Battery-Emulator/issues/627#issuecomment-2593989205

dalathegreat, 2025-03-28:
This ticket can be closed, since https://github.com/dalathegreat/Battery-Emulator/pull/923 is already now implemented in main

closed 2025-03-28 by dalathegreat