dalathegreat/Battery-Emulator #832: Can this interrogate Tesla battery with HV contactors open?

Jaywatts, 2025-01-30:
Hi,
I am going to look at a Tesla model 3 battery and would like to connect my Lilygo setup up to the battery to check it for serviceability. 
Can be done without closing the high voltage contactors? 
If this is the case I wont have to worry about the pyro fuse, capacitor and mucking around trying to get the contactor closed.
Thanks heaps.

dalathegreat, 2025-01-30:
Yes see the Wiki, it has information on [how to check the stats on a Tesla battery before purchase](https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Tesla-Model-S-3-X-Y#how-do-i-test-a-battery-before-buying-)

closed 2025-01-30 by dalathegreat

Jaywatts, 2025-01-30:
Thank you very much. I did see that Wiki and thought it was the case however in the linked video the high voltage contactors were also operated.
Thanks for your time.

Cheers,

Jay