dalathegreat/Battery-Emulator #835: SD Logging links to CAN logger and inconsistencies

murraybester, 2025-01-30:
I have been attempting to log my CAN messages through the SD logging function however I have been running into some problems. 
On my bench setup I have a lilygo connected to a CANbus with the log CAN to SD defined. I have the same setup on a boat with a slightly older lilygo (both are v1.1 however the date on the older one is from 2024-4-17, whereas the one in the lab is 2024-4-25). On my bench setup I have been able to see the SD card initialize correctly using the web based debugging (as well as serial debugging) and can log the CAN messages with ease using the SD log, and CAN logger/export to .txt button on the webserver, however on the boat lilygo I have been having some problems.
Namely, the SD card will not initialize correctly (I have seen the failure message on the web log for debugging, I have not been able to use the serial debugger since I am connecting to this lilygo remotely) and when I try to use the webserver CAN logger the export to .txt button does not work and comes up with an error 404 not found when clicked. Regarding the SD card I have formatted it the same way for both the bench and boat setup (FAT32, both are 32GB brand new cards). The uploaded code is also the same between the 2 lilygos besides wifi ssid and password changes.

Could I get pointed in a direction to try a new approach to logging the CAN on the boat setup or maybe a reason as to why I can't get the SD card to initialize correctly on the boat. I am going to attempt to debug the CAN data using USB serial tomorrow if all else fails.

Additionally, I noticed on my bench setup that when the SD card is not initialized correctly the CAN logger will not let me export to .txt. Is the CAN logger on the web portal specifically linked to the CAN to SD logger? Or am I missing something crucial here...
Any help would be greatly appreciated, thank you!

No-Signal, 2025-01-30:
Could you attach your USER_SETTINGS.h and USER_SETTINGS.cpp files to aid in debugging this problem?

When logging CAN messages to SD is enabled the export to .txt button switches from exporting the CAN web logs to the file on the SD card. As the SD card is not initializing this is then causing the error message.

I have just added a new note to the wiki which is if the SD card fails to initialize for any reason you need to remove power to the battery emulator for a couple of seconds then reconnect power to reset the error, just rebooting or even reflashing will not clear the error so it may be worth trying this a couple of times while I continue to investigate the root cause.

murraybester, 2025-01-31:
Sure I will attach these files. 

If I were to disable to log to SD function and just use the CAN logger on the web portal would it then store and save the log as a .txt file? The CANbus I am trying to log is unfortunately in an area where the signal is not the best, which is why I initially attempted to use the SD logging.

Seeing that the .h and .cpp file types cannot be added to this post I will just copy and paste the code below.
**.cpp:**
```
#include "USER_SETTINGS.h"
#include <string>
#include "USER_SECRETS.h"
#include "src/devboard/hal/hal.h"

/* This file contains all the battery settings and limits */
/* They can be defined here, or later on in the WebUI */
/* Most important is to select which CAN interface each component is connected to */
/* 
CAN_NATIVE = Native CAN port on the LilyGo & Stark hardware
CANFD_NATIVE = Native CANFD port on the Stark CMR hardware
CAN_ADDON_MCP2515 = Add-on CAN MCP2515 connected to GPIO pins
CANFD_ADDON_MCP2518 = Add-on CAN-FD MCP2518 connected to GPIO pins
*/

volatile CAN_Configuration can_config = {
    .battery = CAN_NATIVE,   // Which CAN is your battery connected to?
    .inverter = CAN_NATIVE,  // Which CAN is your inverter connected to? (No need to configure incase you use RS485)
    .battery_double = CAN_ADDON_MCP2515,  // (OPTIONAL) Which CAN is your second battery connected to?
    .charger = CAN_NATIVE,                // (OPTIONAL) Which CAN is your charger connected to?
    .motor = CAN_NATIVE,
    .solar = CAN_NATIVE                   // (OPTIONAL) Which CAN is your shunt connected to?
};

std::string ssid = WIFI_SSID;             // Set in USER_SECRETS.h
std::string password = WIFI_PASSWORD;     // Set in USER_SECRETS.h
const char* ssidAP = "";  // Maximum of 63 characters, also used for device name on web interface
// const char* ssidAP = "";  // Maximum of 63 characters, also used for device name on web interface
const char* passwordAP = AP_PASSWORD;     // Set in USER_SECRETS.h
const uint8_t wifi_channel = 0;           // Set to 0 for automatic channel selection

#ifdef WEBSERVER
const char* http_username = HTTP_USERNAME;  // Set in USER_SECRETS.h
const char* http_password = HTTP_PASSWORD;  // Set in USER_SECRETS.h
// Set your Static IP address. Only used incase WIFICONFIG is set in USER_SETTINGS.h
IPAddress local_IP(192, 168, 10, 150);
IPAddress gateway(192, 168, 10, 1);
IPAddress subnet(255, 255, 255, 0);
#endif  // WEBSERVER

// MQTT
#ifdef MQTT
const char* mqtt_user = MQTT_USER;          // Set in USER_SECRETS.h
const char* mqtt_password = MQTT_PASSWORD;  // Set in USER_SECRETS.h
#ifdef MQTT_MANUAL_TOPIC_OBJECT_NAME
const char* mqtt_topic_name =
    "TestVehicle";  // Custom MQTT topic name. Previously, the name was automatically set to "battery-emulator_esp32-XXXXXX"
const char* mqtt_object_id_prefix =
    "te_";  // Custom prefix for MQTT object ID. Previously, the prefix was automatically set to "esp32-XXXXXX_"
const char* mqtt_device_name =
    "Telemetry1";  // Custom device name in Home Assistant. Previously, the name was automatically set to "BatteryEmulator_esp32-XXXXXX"
const char* ha_device_id =
    "battery-emulator";  // Custom device ID in Home Assistant. Previously, the ID was always "battery-emulator"
// const char* telemetry_topic = "TestVehicle"
// const char* telemetry    
#endif                   // MQTT_MANUAL_TOPIC_OBJECT_NAME
#endif                   // USE_MQTT

#ifdef EQUIPMENT_STOP_BUTTON
// Equipment stop button behavior. Use NC button for safety reasons.
//LATCHING_SWITCH  - Normally closed (NC), latching switch. When pressed it activates e-stop
//MOMENTARY_SWITCH - Short press to activate e-stop, long 15s press to deactivate. E-stop is persistent between reboots
volatile STOP_BUTTON_BEHAVIOR equipment_stop_behavior = LATCHING_SWITCH;
[... 237 lines cut ...]
extern volatile float charger_setpoint_HV_VDC;
extern volatile float charger_setpoint_HV_IDC;
extern volatile float charger_setpoint_HV_IDC_END;
extern volatile float CHARGER_SET_HV;
extern volatile float CHARGER_MAX_HV;
extern volatile float CHARGER_MIN_HV;
extern volatile float CHARGER_MAX_POWER;
extern volatile float CHARGER_MAX_A;
extern volatile float CHARGER_END_A;
extern bool charger_HV_enabled;
extern bool charger_aux12V_enabled;

#ifdef EQUIPMENT_STOP_BUTTON
typedef enum { LATCHING_SWITCH = 0, MOMENTARY_SWITCH = 1 } STOP_BUTTON_BEHAVIOR;
extern volatile STOP_BUTTON_BEHAVIOR equipment_stop_behavior;
#endif

#ifdef WIFICONFIG
extern IPAddress local_IP;
extern IPAddress gateway;
extern IPAddress subnet;
#endif

#endif  // __USER_SETTINGS_H__
```

dalathegreat, 2025-02-02:
The newly released v8.4.0 contains a warning event, to let the user know if SD card failed to initialize https://github.com/dalathegreat/Battery-Emulator/pull/843

murraybester, 2025-02-04:
Thank you for this update. Using the debug via webserver definition I have been able to see the SD card fail to initialize on the boat. I realized the SD card I was using had read-only properties. I decided to get another SD card and attempted to write to a .txt file within it. I successfully did this however this was using a microSD adapter on my computer. I was made aware that an SD card (when using an adapter) can be locked to read-only using the lock switch on the adapter. Is there any chance that the microSD reader on the lilygo could be accidentally put into a locked state? I'm assuming not however at this point I can't understand why the initialization would fail on my boat lilygo but be successful on my bench lilygo. They both use the exact same code and HW...

closed 2025-05-05 by dalathegreat