dalathegreat/Battery-Emulator #838: maxus Batter

0maty01, 2025-01-31:
Hello, I want to ask if anyone has experience of running a maxuse battery

dalathegreat, 2025-01-31:
Maxus is not in the code yet, and no info is available on these batteries. If you have some info, or want to assist, checkout the [wiki page for new integrations](https://github.com/dalathegreat/Battery-Emulator/wiki/Data-needed-for-new-battery-integration)

miraxm, 2025-02-01:
Thank you for answer, with 0maty01 we want to run the battery emulator for 56kwh battery from MAXUS EV80 2019. The battery is out the car and we will try to turn it ON soon. But it is very heavy (670kg) so manipulation is complicated. 
If we will not able to use it's own BMS, we will use external one. 
Would be fine if someone will share his steps with maxus battery if exists some.

M4GNV5, 2025-03-22:
I guess this can be closed as a duplicate of #70?

miraxm, 2025-04-20:
Hello, free days gived me time to try to connect the CAN reader and log the battery bus. I used the USBCAN PCAN clone device and the Savvy CAN software. I saved the log file, but from this point. I am not sure how to continue now. It is very easy to connect this battery, only few wires are there. 
+12V
GND
CAN H
CAN L 
High voltage interlock A
High voltage interlock B 

Not sure about the interlock function - if I make a loop - nothing happen. 

Battery is silent all the time - no relay get ON of OFF. 

Is anybody interesting to help with this? I can leave computer connected to the battery for long weeks if someone would like send some commands to battery. Internet is on site. 

I have another CAN adapter USBCAN-I Pro V502 but I dont know how to connect it to Savvy. 

Lyligo ordered but I dont have Battery Emulator build yet. 

[image]

[image]

[image]

[ev80_2.csv](https://github.com/user-attachments/files/19824483/ev80_2.csv)

AntonyLeons, 2025-04-29:
hey @miraxm could i get the pinouts from the low volate connector, and a picture to help.

miraxm, 2025-04-30:
Yes, it Is very easy.. but what battery do you have?