dalathegreat/Battery-Emulator #839: Afore RESET_PANIC

eham365, 2025-01-31:
RESET_PANIC using Afore inverter, Afore CAN protocol and Leaf Battery. 

The system has been up and running for 7 days in off-grid mode with only the battery connected. Today the PV got connected to the inverter and after 3kWh of charge I got this reset.  

[log_10-18-39.txt](https://github.com/user-attachments/files/18624299/log_10-18-39.txt)
[image]
[image]
[USER_SETTINGS.ccp.txt](https://github.com/user-attachments/files/18624300/USER_SETTINGS.ccp.txt)

dalathegreat, 2025-01-31:
I took a look at the Afore protocol, but I could not find any obvious things like division by 0 or something that would cause a crash.

But, I noticed that you are using a .dev version of 8.1, would it be possible for you to update to the latest stable (at the moment 8.3.0). Reason for this is that it is extremely difficult to know exactly the .dev state when the code was compiled.

If the issue continues to happen with a stable version, we can look more into it!

eham365, 2025-01-31:
Thank you for the quick replay. I will update to the latest version tonight.

closed 2025-02-16 by dalathegreat