dalathegreat/Battery-Emulator #846: Tesla Model 3 battery pre purchase inspection

Jaywatts, 2025-02-03:
Hi, I'm not sure if this is the correct place to ask this question, if not please point me in the right direction.
I have made a test box with a LilyGo running the latest battery emulator and am planning on testing a model 3 battery before purchase.
The car had done 30000ks and was at 36% when the battery was removed just over a year ago.
Just wondering if anyone could tell me what cell voltage or cell voltage split would be considered unacceptable.
Thanks for your time 😊

dalathegreat, 2025-02-03:
Ideally mV diff below 100mV high-low. Lower diff the better. 

UI will turn yellow incase the diff is too large, and warn you with an Event, as explained in the Wiki https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Tesla-Model-S-3-X-Y#how-do-i-test-a-battery-before-buying-

WilliamKTurner, 2025-03-23:
Do you need to connect 12V to the PCS studs to get the battery to talk CAN to the LilyGo, or is that only required if you want to try and get the contactors to close?
For a pre-purchase inspection, I don't think I will have time to install all the bridges required to get the contactors to close, so I just want to talk to the pack to check the cell voltages and the total charge/discharge energy.

dalathegreat, 2025-03-23:
@WilliamKTurner See the wiki: https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Tesla-Model-S-3-X-Y#how-do-i-test-a-battery-before-buying-

See example: https://secondlifestorage.com/index.php?threads/brisbane-full-backup-tesla-model-3-home-battery-project.12987/#post-93398

closed 2025-03-23 by dalathegreat