dalathegreat/Battery-Emulator #852: RJXZS - unreliable SOC when system is idling

pazdzierzz, 2025-02-04:
**What**

Correction of battery SOC during longer idles. 

**Why**

RJXZS does not care about total pack voltage and temperature when estimating SOC. Hall sensor also does not detect small parasitic battery drain caused by inverter. That causes a situation when battery voltage is dropping with SOC freezed (estimated 0,01V drop per 3 hours with Sofar inverter). After a while that causes battery contactors to open because of cell discharge limit protection. 

**How**

BMS calculates SOC (battery capacity percentage) based on two values: battery pack capacity and battery usage capacity

SOC = (bp capacity-bp usage capacity)/bp capacity * 100%

[image]

[image]

Maybe there is a way to create a lookup table of cell voltages for Lily - when total voltage of battery will be not in line with voltage estimated SOC , Lily will send battery usage capacity correction to BMS. 

Some provisional table created with ChatGPT (surely have to be verified)
[image]

CAN ID for BMS to receive battery used capacity:

[image]

pazdzierzz, 2025-04-13:
Bumping topic a little - with a small update.

RJXZS can automacitally recalibrate SOC to 100% when upper voltage limit is reached (Auto Clear Cap function)

So I'd be nice to take care of the other side of voltage, when battery is discharged/idles for a longer period

dalathegreat, 2025-04-13:
Should we add this info to the Wiki page for RJXZS? After that we can close this issue?

pazdzierzz, 2025-04-14:
Yes, I'd be nice to add that it's good to charge up battery to its limit to recalibrate SOC

pazdzierzz, 2025-04-14:
Just updated RJXZS wiki page, also added important notification about events stored in BMS which have to be cleared before first startup.

closed 2025-04-14 by dalathegreat