dalathegreat/Battery-Emulator #853: Can rx failure

ougueur, 2025-02-04:
Good morning
,I have had this fault for some time, no more communication with the battery. I opened the battery and removed the BMS to eliminate all the connection problems. Still no CAN communication
.Termination resistor OK
2,4 v sur can h et can I 
No communication with leafspy either
.The battery capacity of the Nissan Leaf is 30 kWh.can we have communication Canonly by powering the calculator?What other checks are performed before BMS replacement or other solution for reuse of the battery pack?

[image]

[image]
[image]
[image]

Sincerely

dalathegreat, 2025-02-04:
Looks like the BMS is completely broken yes. Maybe best to start looking for a new BMS! We have had several users with broken LEAF BMS after [lightning strikes](https://github.com/dalathegreat/Battery-Emulator/wiki/Lightning-strike) in the area

ougueur, 2025-02-04:
Ok... Do I have to take exactly the same reference for the BMS?
Is an alternative BMS possible?
Not lightning strikes...but a  rapid restart is the cause of the problem I think...

dalathegreat, 2025-02-04:
Best to fit a 30kWh BMS, if you try to use a 24kWh, you have to change lots of wiring and the current sensor is different.

ougueur, 2025-02-18:
Ok, I can't find a bms with exactly the same reference ...
293a04nr6a
I saw several bms that are supposed to come from the leaf 30kwh .. but the reference is different ...
Are there several compatible references?

If I can't find it, I had in mind to use a BYD hvm / s bms, but I can't find the pinout
or the information on the signals necessary to send to the BYD bms ... I saw that you have already repaired byd batteries, do you have documentation?

[image]

dalathegreat, 2025-02-19:
Any 30kWh BMS will work.

I dont have the BYD wiring, sorry

closed 2025-05-05 by dalathegreat