dalathegreat/Battery-Emulator #855: Sofar HYD10 ktl 3ph, tesla model y battery and bts 5k

EliasVP, 2025-02-04:
Hello, 

Ik have a Sofar hyd10 ktl 3ph inverter a pair of bts 5k batteries and a tesla model y battery (which I have all hooked up, tested and ready to go).  But while I was testing I came to the conclusion that my inverter (which has 2 battery inputs) can only have 1 type of battery connected. So if the battery that's connected to input 1 is a pylontech then the battery on input 2 also has to be a pylontech. Since I already have 2 bts 5k batteries, which I would like to continue to use, I would have to emulate the tesla battery as a bts 5k or I would have to emulate the bts 5k as a pylontech. 

I have seen that the sofar can is under development, is this to emulate the bts 5k? Or is there another way to solve my problem?

Regards,
Elias VP

pazdzierzz, 2025-02-05:
As far as I know BTS use DC-DC converter, It'd be hard to emulate it.

EliasVP, 2025-02-10:
Yes, the batteries themselves are 50 odd volts, and I think the BDU (battery distribution unit) then takes it to 400v. But I believe  it's also the BDU that communicates to the inverter, so I don't know if that whould be an issue.

pyeco64, 2025-03-06:
Hello.

Can you share setting? Having problems with inverter comunicatino. Sofar HYD20KTL 3PH.

GD

EliasVP, 2025-03-06:
With what are you having comunication problems? 
With a lilygo or with a bts 5k?

dalathegreat, 2025-05-25:
Workaround is to use 2x emulators, and convert the OEM Pylon battery into BYD CAN. We have the option for this in the software.

Example, Tesla battery + Chinese Rosen battery, on a double input Deye inverter. Both emulating BYD CAN.
[image]

If the inverter needs Pylon, just have the second battery be a Pylon SEND_1.

This is all very doable! Closing this issue!

closed 2025-05-25 by dalathegreat

pazdzierzz, 2025-07-11:
@EliasVP  - SOFAR Can is now operational (check pull requests), so you can try to use that protocol for connecting tesla battery. Should work with BTS. Tesla pack should have different address set.

EliasVP, 2025-07-11:
@pazdzierzz That's amazing, I'll see if I have some spare time this weekend. If so I'll definitely give it a try

dalathegreat, 2025-07-11:
Wiki has been updated, feel free to improve the Wiki you two incase you have more info than me :) https://github.com/dalathegreat/Battery-Emulator/wiki/Inverter:-Sofar#sofar-can