dalathegreat/Battery-Emulator #881: New battery integration, EMUS G1 bms

ErikssonPer, 2025-02-11:
I've made an attempt at creating an integration for an EMUS G1 bms in growatt lv can mode.
I'm running it on my system right now with a 100cell battery and it works for the most basic stuff like voltages, amps, temps, soc and soh

Error handling is still not figured out.

I'm no programmer by any means, it's mosly copy pasted from other integrations so I'd like someone to give it a look over before I go further and maybe it can be integrated here.

The reason it's in growatt lv can mode is that's the only protocol it has that I can get the data needed.
It has several other can protocols it can emulate but I don't have spec for those.

dalathegreat, 2025-02-12:
Feel free to add a Draft PR and we can review it! 👍

ErikssonPer, 2025-05-25:
Oh, forgot to update, In the new version of EMUS software they released support for Pylon HV protocol so I just ended up using that since it's already implemented here.

closed 2025-05-25 by ErikssonPer