dalathegreat/Battery-Emulator #894: BYD Atto 3 - Contactor Control / Cell Monitor

demon1300, 2025-02-15:
Hi All,

Firstly - love this project, THANKYOU to Dala and all of the contributors thus far.

I'm just in the process of connecting a BYD Atto 3 to my Fronius Symo Gen24 Plus and have been connected to the battery first in the trailer to make sure everything there was fine.  I was lucky and managed to get a non crashed Atto 3 pack, so everything works fine on powerup without having to delve into the battery to get the contactor and pre-charge relay working... Phew!

My question/feature request (as i couldn't find any info anywhere on these two areas for this specific battery);

1. Is it possible to get the contactor control working in the GUI for the BYD Atto 3 battery over CAN?  Would be a great safety addition to allow the software to open the contactors if any of the safety thresholds are met.  Assume it has some form of control now as the contacts close when the Lilygo is powered on?
2. Is the cell monitor suppost to work for this battery?  At the moment it's just blank for my pack and wanting to know if this is normal?

Just tried V 8.6.0 and can confirm the temp info is now displaying also in the more battery info page, so thanks for adding/fixing that!

Thanks!

Rob

dalathegreat, 2025-02-15:
Hi @demon1300 

1. Yes this functionality is missing, and could be added with some development. CAN log from a vehicle that is turning off is needed to get this working.
2. This can also be added, but a CAN log from an OBD2 dongle reading cellvoltages from a vehicle is needed to get this working

Enjoy the non-crashed battery! 🙌

demon1300, 2025-02-22:
Hi @dalathegreat thanks for the reply.  I've just had a look at a quick log that I've caputred via USB from the Lilygo - connected to the battery I have

Based on some quick analysis each 0x43D message contains three cell voltages in millivolts (mV), encoded in Bytes 1-2, 3-4, and 5-6, with Byte 0 as a counter. The counter runs from 00 to 29 (hex), suggesting 42 messages covering 126 cells (42 × 3 = 126).  This is precisely the cell count of this 60kw battery.  The output contained for all 126 of these cells in my log matches exactly to what the SOC reads in the emulator, with the total deviation also being 5mv (lucky me!).

Sample:
(162.50) RX0 43D [8] 00 D5 0C D5 0C D5 0C 00
(162.60) RX0 43D [8] 01 D5 0C D4 0C D4 0C 00
(162.70) RX0 43D [8] 02 D5 0C D6 0C D7 0C 00
(162.80) RX0 43D [8] 03 D6 0C D6 0C D6 0C 00
(163.00) RX0 43D [8] 05 D6 0C D7 0C D6 0C 00
(163.58) RX0 43D [8] 0B D8 0C D7 0C D7 0C 00

Breakdown:
1. Byte 0: Increases sequentially (00, 01, 02, 03, ...), likely a frame counter or cell group identifier.
2. Bytes 1-6: Pairs like D5 0C, D4 0C, D6 0C, D7 0C. These could be voltage readings in big-endian format (e.g., 0x0CD5 = 3285 decimal).
3. Byte 7: Usually 00, possibly a status byte or padding?

Converting to Voltages:
D5 0C (hex) = 0x0CD5 = 3285 decimal = 3.285V
D4 0C (hex) = 0x0CD4 = 3284 decimal = 3.284V
D6 0C (hex) = 0x0CD6 = 3286 decimal = 3.286V
D7 0C (hex) = 0x0CD7 = 3287 decimal = 3.287V
D8 0C (hex) = 0x0CD8 = 3288 decimal = 3.288V

Would it be possible for you to implement this into the individual cell monitor code as per my original point 2 question?

Cheers and keep up the good work!

Rob

dalathegreat, 2025-02-22:
Excellent work Rob! I have implemented this in this PR: https://github.com/dalathegreat/Battery-Emulator/pull/911

You can test it via this direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/byd-atto3-cellvoltage-reading.zip

demon1300, 2025-02-22:
> Excellent work Rob! I have implemented this in this PR: [#911](https://github.com/dalathegreat/Battery-Emulator/pull/911)
> 
> You can test it via this direct download link: https://github.com/dalathegreat/Battery-Emulator/archive/refs/heads/feature/byd-atto3-cellvoltage-reading.zip

@dalathegreat Can confirm it works perfectly!!  Thanks for implementing that so quickly!

Now to turn my attention to the contactor opening and closing to see if we can get that working via the GUI...

[image]

closed 2025-05-05 by dalathegreat

2thgys9vmm-sudo, 2026-10-25:
[image]
_

2thgys9vmm-sudo, 2026-10-25:
Problem solving?

dalathegreat, 2026-10-25:
@2thgys9vmm-sudo , please create a new Discussions thread, if you cannot get the recovery undercharged mode to work https://dalathegreat.github.io/Battery-Emulator-Wiki/setup/hardware/recovering_undercharged_battery/