dalathegreat/Battery-Emulator #900: 30kWh Leaf Pack - Reset Degradation Data

malcolmputer, 2025-02-18:
When combined with the daily reset feature or the BMS reset via MQTT or power cycling the entire setup during grid-loss, the 30kWh pack resets to the previously stored SoH.  I seem to recall a conversation that BennVenn said it was related to the battery needing to see some messages from the "rest of the car" to make the new SoH stick. 

Maybe we have an option for reset anytime SoH isn't 100% for a 30kWh pack so the BMS power resets don't clear that and require manual intervention? Or we could have a MQTT endpoint for reset SoH.

In any case we need to make sure the solution works for double-battery as well, to prevent another case like Issue [899](https://github.com/dalathegreat/Battery-Emulator/issues/899)

Micr0mega, 2025-11-03:
Great! I have simulated what you described, and I can confirm the reset SOH sticks after following these steps manually:

1. Open contactors
2. Reset battery degradation
3. Disconnect CAN cables (I also disconnected the inverter cable as they share a single CAN bus)
4. Wait 3 minutes
5. Reconnect CAN cables
6. Close contactors
7. Reboot Battery Emulator

After these steps, the current SOH stays 100%, even after another reboot 👍

dalathegreat, 2025-11-03:
Superb! I will close this issue, added instructions to the Wiki on how to perform the SOH reset properly! https://github.com/dalathegreat/Battery-Emulator/wiki/Battery:-Nissan-LEAF---e%E2%80%90NV200#performing-the-reset-in-detail

Massive thanks for figuring these steps out @Micr0mega 🙌