dalathegreat/Battery-Emulator #901: [BUG] OTA page lost return to battery-emulator buttons in 8.6.0

marknisk, 2025-02-18:
After 8.6.0 OTA page does not have buttons that can be used to return to Battery-Emulator main view.

**To Reproduce**
a) hit "OTA Update" in the main page
b) after 8.6.0 is flashed the user sees OTA page without buttons

**Expected behavior**
Buttons that make it easy to navigate back to Battery-Emulator main page

**Screenshots**

[image]

**Version and settings (please complete the following information):**
- Software version: 8.6.0
- Battery used: NISSAN_LEAF
- Inverter communication protocol: BYD_MODBUS
- Hardware used for Battery-Emulator: HW_LILYGO
- CONTACTOR_CONTROL: yes
- MQTT: 

[image]

yes

dalathegreat, 2025-02-19:
Yes this customization is what prompted us to release 8.6.0, the ESP32 3.1.2 update caused this page to crash. By reverting our customization we got it going again. We need to look at how to re-implement it in a way that does not cause crashes. @amarofarinha might be able to solve this once they have time over!

dalathegreat, 2025-02-19:
ESP32 v3.1.3 is now releaesed, which fixes the ArduinoJSON string copy crash. I have reverted the removal of the OTA buttons in this PR https://github.com/dalathegreat/Battery-Emulator/pull/902

dalathegreat, 2025-02-20:
Fix now in main. Version 8.7.0 will restore the OTA page functionality!

closed 2025-02-20 by dalathegreat