dalathegreat/Battery-Emulator #907: Tesla model y battery stops charging 12v battery

EliasVP, 2025-02-21:
Hello,

I have a tesla model y battery and I have it connected the same way as the scheme on the tesla model 3/y page (see the photo below). It works completely exempt that when the voltage on the pcs studs reaches 14,1v it stops outputting voltage. And then my 12v battery slowly drains (because the power usage of the bms and the lily-go). I figure that it's normal for it to stop charging when it reaches 14v because that's the voltage the original battery is supposed to be charged to. And because there's a diode (D2 and 3) between the battery and the pcs it can't measure the voltage of the battery and then just won't restart charging cause it's like there's no 12v battery. Only when I close the startup switch for 2 seconds or so it restart the charging. 

Is it normal fot it to act like that or is there something I overlooked?

Would placing a capacitor on the pcs studs be a way of overcoming this problem? I figure that unless the pcs draws power after the contactors have closed, the capacitor will drain as the battery's voltage drops and when low enough, the pcs will restart the charging. 

[image]
[image]

DeadmaroZ-TLOTL, 2025-02-22:
The same problem with a few batteries!

EliasVP, 2025-02-22:
Thanks, I'll look for a way to work around it.

Gangster8214, 2025-02-25:
I had this problem on two batteries when using a custom 4s2p Li-ion battery instead of a lead acid. Although I had previously used a Li-ion on one battery successfully. The problem was solved by replacing the battery with an AGM 9AH lead acid and changing its wires to thicker 2.5 sq. I never used any diodes between the battery and the PCS. The voltage was never more than 14.2v.

[image]

DeadmaroZ-TLOTL, 2025-02-26:
That’s not true; I have batteries that work with AGM and LFP S4, and I have batteries that don’t work with either AGM or LFP. I think the internal resistance is being measured there.

EliasVP, 2025-02-26:
So I tried connecting a capacitor to the pcs studs and this fixed the problem, the 12v battery continued to be charged, even when the pcs voltage was 14,2V it didn't shut off. But once the 12v battery was close to full and the current that was being drawn was getting smaller and the voltage was getting closer to the 14,2V, I started to hear a high pitched sound which became louder when the current was getting smaller. So if checked the signal of the pcs studs with my ocilloscope and there was a small ripple on the 14V. So i think that the pcs was constantly stopping and then again starting to uitput voltage which resulted in the sound I was hearing. 

So I just went and bought a this power supply. It has an adjustable output voltage (11-14v) so I could simply set it to 12,7V to keep the 12v battery topped up and supply the bms. So now the battery is completely working like it should 

[image]

EliasVP, 2025-03-23:
Hi, I'm sorry it took so long for me to get back at you, but here are the can logs 

[image]
[image]
[image]

DeadmaroZ-TLOTL, 2025-03-23:
All problem in long wiring, tesla battery mesure internal resistance.

EliasVP, 2025-03-23:
Oh really.  I'm glad he found his problem.

1ni3shi, 2025-03-24:
Thank you for the can logs, another guy had provided his and I was able to get my battery working.
But it's good to know that both of you have the same 20A can ID, good verification.