dalathegreat/Battery-Emulator #908: [Feature request] More configurable LED options

TommySharpNZ, 2025-02-21:
I would like to be able to use addressable LEDs mounted on or close to that battery that allow me to visualise the state of the battery such as charge/discharge, battery issues etc.

I have a Stark CMR which does have available gpio outputs.

We should consider a strip of LEDs but in my case I want a ring of LEDs that remind me of Tony Stark's Arc Reactor.

- White : Discharging
- Blue : Charging
- Red : Battery Issue
- Orange : Disabled/Off
- Effects : Pulse or Chase with speed reflecting charge/discharge speed

Obviously colours and effects would be configurable.

I know this is mostly just eye candy and could be done using a separate ESP32 but having it integrated would mean more responsive and also less points of failure.

dalathegreat, 2025-02-25:
First step of this added: https://github.com/dalathegreat/Battery-Emulator/pull/916

dalathegreat, 2026-04-06:
Closing this as completed, since v9 LED patterns are configurable in Webserver

closed 2026-04-06 by dalathegreat